iNavFlight/inav #3909: Matek f405 OSD INAV 2.0 regular plane servo mixer (same as #2912)

zingaros, 2018-10-07:
Hello

it looks like on 2.0 for target matek F405 osd, using a third servo is not possible.
Works on 1.9 for pad S5. On 2.0 I did not suceed to use S4 S5 or S6 to drive a servo.
After looking to git hub, looks like the change referenced here #2912 was not moved to 2.0.

I would have loved to fix myself but not not feel confident into building the required dev env.

shellixyz, 2018-10-08:
It has been disabled again following #3224. You will have to enable it yourself.

zingaros, 2018-10-08:
So it means per Matek request f405osd is for multi rotor or wings (2 servo) and only that?
@mateksys it does not make sens

MATEKSYS, 2018-10-08:
I am confused. LED pin should be on S5(PA15),  I don't know when it was changed.

zingaros, 2018-10-08:
@mateksys what I m looking for is that as for 1.8 and 1.9 I can use the matekosd to drive more than 2 servo for use in a plane. Looks that with the change in 2.0 we can only drive 2 servo. It’s fine for a wing but not anymore for a plane.

MATEKSYS, 2018-10-08:
@zingaros  not really sure what you mean.  with 1.8/1.9/2.0.  S1 is for motor,  S2/S3/S4/S6 are for fixed wing servos, no changes.

zingaros, 2018-10-08:
In my experience S6 does not work to drive a servo neither S4 with the Matekf405 osd target on INAV 2.0.
Am I doing it wrong?

MATEKSYS, 2018-10-08:
no difference on PWM output definitions between MATEKF405 & MATEKF405OSD target. 
CLI "defaults" after reflashing it with 2.0

shellixyz, 2018-10-08:
@MATEKSYS I don't know why DE changed it again in dev. I can revert it.

zingaros, 2018-10-14:
For other users that might face the same trouble:
I confirm that despite test, re-test , reset re-config etc, no way to have more than 2 servo for wing with inav 2.0 on a matek f405osd. I this a way to sell the more shiny F405 wing?
I'm not a dev but thanks to the good doc, I was able to build the dev env, change the code myself (see change here #2912) in a local repo and build an image.

closed 2018-12-27 by stale[bot]