iNavFlight/inav #3911: Wrong readings on iNav with digital airspeed sensor

dovcharov, 2018-10-08:
Good day, iNav development team. Thank you very much for your efforts and wonderful firmware.

I decided to install an airspeed sensor (pitot tube) on my Reptile S1100 flying wing. The wing is equipped with Matek F722 flight controller. I picked a px4-compartible digital airspeed sensor (with i2c interface) from [here]https://hz.aliexpress.com/item/Ardupilot-Arduplane-Pitot-Digital-Air-Speed-Meter-Airspeed-Meter-Sensor-Tube-for-Pixhawk-PX4-Flight-Controller/32560997823.html(url). Today I installed pitot tube  into the wing, connected SCL and SDA pins on airspped sensor to the flight controller. I added the airspeed indicator on the OSD. 
The firmware detected the airspeed sensor, and the "Pitot" icon in iNav configurator turned blue. 
However, I get the wrong readings in iNav OSD. In still state (in the room) OSD displays 426 km/h of air speed. Once I start blowing into the pitot tube, the speed on the OSD goes down - to 405-400-390 km/h. So I get some readings from the airspeed sensor, but they seem to be inverted. 

What am I doing wrong? Are there any parameters that I can play with to tune the sensor?

Thanks in advance for your support, 
Best regards.

digitalentity, 2018-10-08:
It's not the first report of the issue with digital airspeed sensor. Unfortunately I don't have any of those to troubleshoot. I guess it's time for me to get one.

dovcharov, 2018-12-23:
Thanks for reopening. The issue is still there

dovcharov, 2019-02-02:
I no longer have false reading of 425 km/h with inav 2.1 rc1. When wing is on my desk, i see 1-2 km/h on my osd. I will check readings in flight tomorrow.

dovcharov, 2019-02-10:
Airspeed sensor works now. Please close the issue.

closed 2019-02-10 by dovcharov