iNavFlight/inav #3913: INAV 2.0 Matek F411 FC Disarming in Flight!

JDFPV, 2018-10-08:
INAV 2.0 Matek F411 FC Disarming in Flight! Mini AR Wing

 Using INAV 2.0 on a Matek F411 FC. Default settings for the 600mm wing presets. I also followed the steps to enter the commands in the cli from the inav wiki fixed wing page. So after launch around 4 minutes into flight i was feeling everything was going well... Then all or a sudden i see the stat screen and have 0 control over the plane. So i flipped the disarm switch, lowered the throttle and flipped the switch to arm and i regain control. So i was puzzled thinking my radio signal had dropped out and some how failsafe didn't kick in.. Which i had confirmed that failsafe did work when i tested it on the bench. So i kept flying just to see if it would happen again. 6 minutes later the stat screen popped back up again. So i repeated the step as mentioned before to regain control and i could not get any response out of the FC causing my wing to crash. So i took it home thinking the rx for radio control went bad and installed a new one. And i went over all the setting in inav. Everything looked ok. Next day, same thing happened again but i regain controlled and landed. Bug possibly or glitch?? Video link below of osd footage of flight.

https://youtu.be/KScxmxyXUL4

 # dump

# version
# INAV/MATEKF411 2.0.0 Aug 20 2018 / 19:23:35 (dbdd1656a)

# resources

# mixer

mmix reset

mmix 0 1.000 0.000 0.000 0.000
mmix 1 1.000 0.000 0.000 0.000

# servo mix
smix reset

smix 0 3 0 50 0
smix 1 3 1 50 0
smix 2 4 0 -50 0
smix 3 4 1 50 0

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 -100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
[... 412 lines cut ...]
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 40
set pitch_rate = 15
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 250
set vbat_warning_cell_voltage = 275
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

#

teckel12, 2018-10-08:
@JDFPV The video says that you disarmed by SWITCH.

digitalentity, 2018-10-08:
@JDFPV what radio system are you using? Some radio systems (most notable FrSky R9) are known to send random channel values on some occasions. 

INAV 2.0 uses a lot of fixes to make RC link more reliable, but you should not rely on firmware to fix otherwise broken hardware.. You can try setting `sbus_sync_interval = 1500`, it will make INAV work correctly with receivers violating SBUS standard. Disabling `disarm_kill_switch` is also a good idea to make sure you don't accidentally disarm while throttle is above zero. 

From video this looks like genuine switch disarm - my guess is that it's a flaky radio hardware.

JDFPV, 2018-10-08:
i'm using dragonlink v2 on a devo 10 radio. Been using it for years with 0 problem. And actually i flew my other plane after this and it has the say profile settings same disarm switch and it works fine. But i will try the set disarm_kill_switch = OFF.

digitalentity, 2018-10-08:
It may be a problem with particular receiver. I assume each of your planes have it's own receiver.

JDFPV, 2018-10-08:
Each plane uses a micro Dragonlink. 3 planes have inav 2 has fy41ap lite flight controllers.

JDFPV, 2018-10-08:
All other planes work fine with inav. Using 1.8 on the Nano Goblin. Using 1.9.1 on my fx79.

digitalentity, 2018-10-08:
The only thing we can conclude from your video - FC was disarmed by a switch. Without BB log it's hard to troubleshoot any further. Since MATEK411 doesn't have the onboard flash, using old good OpenLog is the only option here.

JDFPV, 2018-10-08:
And I actually changed the rx on the 2nd flight.

JDFPV, 2018-10-09:
what is OpenLog ?

giacomo892, 2018-10-09:
@JDFPV  https://www.sparkfun.com/products/13712 Serial logging device.

JDFPV, 2018-10-09:
I totally dont have space for it in the plane I'm having trouble with. I will test fly it with the changes made this weekend and I will report back to you guys. Thanks

JDFPV, 2018-10-15:
Problem solved. The internal 5v 3a bec on the brand new matek f411 board went bad. Causing the rx to go nuts with over voltage. Thanks INAV!

closed 2018-10-15 by JDFPV