iNavFlight/inav #3914: Airbot Omnibus F4 V6 won't recognize iBus receiver

CaptainYags, 2018-10-08:
Airbot Omnibus F4 V6 can't get my iBus receiver to be recognized. 
INAV/FIREWORKSV2 2.0.0 Aug 19 2018 / 21:25:18 (dbdd1656a)
Is that the right target that I should be using?

The airbot manual says that the gyro needs to be selected by typing "set gyro_to_use = 1" in betaflight and it has effects on the UARTS, Unfortunately, it doesn't work in iNav. Are there any other CLI commands that I need to use?

digitalentity, 2018-10-08:
There are so many OMNIBUS F4 boards. V3, V4, V5, now V6. All of them have random changes in schematics which make them unsupportable. Also, try using OMNIBUSF4PRO target.

CaptainYags, 2018-10-09:
That's why I purposely picked the Airbot one, since they are on the sponsored boards list! It's really starting to seem impossible to find a well supported board for iNav! What hardware are we supposed to use?!

Tygrys-1, 2018-10-12:
V6 is a new product with 5 UART-s advertised. It is definitly different from the previous design, probbaly sharing much but not all.

teckel12, 2018-10-12:
@CaptainYags Never a good idea to go with new boards. Go with well tested FCs.

Never a problem with real Omnibus F4 Pro. If you need more UARTs the Matek F722 is looking like a good option. But I still haven't switched because I'm not 100% sure it's been proven yet. Will probably attempt a winter build on F722 so I can sort out any unforeseen issues. 

If you go for the latest FCs, always expect headaches, problems & wasting money.

digitalentity, 2018-10-14:
@CaptainYags please post a link to the board (where you purchased it). I'd like to get one and debug

teckel12, 2018-10-14:
@digitalentity I believe it's this board: https://www.unmannedtechshop.co.uk/omnibus-f4-v6-flight-controller/ from what I've gathered from this issue and another one.

CaptainYags, 2018-10-15:
Here is the board:
https://store.myairbot.com/flight-controller/omnibus-f3-f4/omnibusf4v6.html

Running the FIREWORKSV2 target, I got my iBus receiver to work by connecting it to RX1 instead of the SBUS port.  So far everything else seems good too.

CaptainYags, 2018-10-16:
I thought that SBUS and IBUS were both inverted, but perhaps not.  Still ironing out some other issues before I get airborne.  So far haven't had luck with 2 different compasses on SCL/SDA, but I'm using the GPS connector, and will try soldering to the pads tonight

The receiver is working though, thanks for the help. I can open another issue if I have a different problem, but the FIREWORKSV2 target seems like it's on the right track for now

teckel12, 2018-10-16:
@CaptainYags I don't think IBUS is inverted.  Further, IBUS is a two-way protocol unlike SBUS, and ports on F4 FC designed for SBUS almost always have a one-way inverter.  Most F4 FC will have a pad to disable the inverter if you're using any other Rx protocol.  Or, they have a totally different set of pins for the same UART that isn't inverted.  It sounds like the Airbot Omnibus F4 V6 doesn't have either of these.  So, the board is really designed for SBUS only.  Or, if using a different Rx protocol you lose a UART (which kind of negates the value of the board as the number of UARTs is the reason it exists).

In other words, it seems you're going to need to use a different UART to use IBUS on this FC.  If you had a link to more detailed pictures or schematics of the board that may allow for a hardware "fix".  I can't find anything other than lower resolution images of the board.

CaptainYags, 2018-10-19:
I said I got it working. The SBUS pad is the inverted pad for UART1, and RX1 is the non-inverted pad for UART1. The IBUS is working fine on RX1 without any other changes,

It was my confusion. I was under the impression that SBUS and IBUS were identical protocols by different manufacturers, and I've had success connecting the IBUS to SBUS pads on other boards.  I was wrong and just got lucky before.

closed 2018-10-19 by CaptainYags