iNavFlight/inav #3915: Migrate to threaded scheduler

digitalentity, 2018-10-09:
Allow low-priority tasks to yield.

Current scheduler assumes that each task iteration will finish in much less than one gyro/pid cycle time. Full-graphic OSD, some operations on character OSD, complex position estimation/navigation code is quite time-consuming and in some cases will cause significant cycle time jitter. 

Migration to full RTOS preemptive scheduler is likely an overkill, something like coroutines should be enough.

StewLG, 2018-10-21:
I believe I am suffering from just these performance issues with this iNav related work:

https://github.com/StewLG/CraftServices/

I get incomplete frame draws when (for example) too many waypoints are drawn on the radar screen.

digitalentity, 2018-10-28:
Partly done in #3962. Protothreads/coroutines could be seamlessly integrated into current scheduler and eventually replace the scheduler.

Linjieqiang, 2018-12-24:
@digitalentity What about use FreeRTOS with INAV?

digitalentity, 2018-12-24:
@Linjieqiang INAV was not designed to be thread-safe. RTOS will create more problems that it will solve at the moment.

Linjieqiang, 2018-12-24:
Okay. Thanks.

closed 2023-09-29 by DzikuVx