iNavFlight/inav #3916: OMNINXT F7 can't load fonts or read logs

CaptainYags, 2018-10-09:
Airbot OMNINXT F7
iNav version INAV/OMNIBUSF7NXT 2.0.0 Aug 20 2018 / 19:15:39 (dbdd165)

The board/configurator hangs when I do either of:
-Try to download blackbox data
-Try to upload a font for the OSD

CaptainYags, 2018-10-11:
I tried flashing the board to Betaflight, and with Betaflight I can download the blackbox and upload fonts.  After that I flashed back to iNav, and I can't upload/download again.

VladimirShakhov, 2018-10-16:
I confirm the problem - the font in the osd does not load.

Rclay11541, 2018-11-06:
I too confirm. ODD does not function, upon attempting  to load a different  font as a work around, I also can confirm  the FC hangs indefinitely.

digitalentity, 2018-11-06:
What OS are you using? MacOS?

VladimirShakhov, 2018-11-07:
I'm using win7 64bit.

Rclay11541, 2018-11-07:
I'm using Win 7, 64 bit as well.

Further, the new update to INAV 2.1 has caused an error in RTH. Exact same settings and in version 2.0 RTH works excellent. In V2.1 it goes off in a random direction and at times upon engaging RTH shoots way up at max throttle in altitude....other times it's a slow accent.

Rclay11541, 2018-11-07:
Further OSD still doesn't function in v2.1 or v2.0

VladimirShakhov, 2018-11-14:
unfortunately, the osd does not work (((

Moris644, 2018-11-27:
OSD does not work!

digitalentity, 2018-11-27:
Fixed by #4026

closed 2018-11-27 by digitalentity