iNavFlight/inav #3917: Problem with Matek F411 Wing GPS

rwilabee, 2018-10-10:
I am having a problem pictures attached.  I am getting GPS altitude on my OSD screen but the altitude is showing up in the Inav GPS screen.  My speed shows up just fine in my OSD screen.  I have the Baro turned on also since there is one built in to the F411 board.  Any ideas
[image]
[image]

Thanks 

For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)
* [IRC channel](irc://irc.freenode.net/#inavflight)
* [Slack channel](https://publicslack.com/slacks/inavflight/invites/new)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)

DzikuVx, 2018-10-11:
Did you tried tat in fligt too? On a bench it will always be 0 since it's relative, not absolute ASL that is displayed in OSD

closed 2018-10-11 by DzikuVx

rb1205, 2018-10-12:
The altitude you see on the OSD is relative with the reference being continuously set at the current altitude until the craft is armed, so the displayed altitude will always be 0 while it is disarmed

Also consider that AFAIK if you have a barometer then it's altitude replaces the  the signal from de barometer rrplave