iNavFlight/inav #3918: RTH altitude management in fixed wing

igi1, 2018-10-10:
Matek F411 in an electric glider
"nav_rth_alt_mode" set to fixed and 7000

issue:  When I switch to RTH the plane returns allright and climbs or descends on the way back trying to reach the dialed in 70 meters. But once it enters the loitering pattern it keeps the altitude it has when entering the pattern.  In other words it cannot descend or climb once in the RTH loitering circle. This is a major shortcoming if, like in my case, one wants to use RTH to retrieve the plane in case it climbs out of sight in a strong thermal. As is hitting RTH at great hights  will make the plane return but it remains high and might even still climb in the thermal, while circling overhead, because it could not descend sufficiantly to the specified altitude on its way home. 

Checking with a friend who uses the V 1.9.1 we found that in that version things work as they should , that is the plane keeps descending to the specified altitude even after reaching 
the loitering pattern. 

Please revert to that RTH behavior in V 2.0

b14ckyy, 2018-10-23:
I can confirm that. Plane will keep the altitude that it had, when entering the RTH Loiter instead climb or dive to the set altitude.

wx4cb, 2018-10-31:
i can also confirm that. i was flying at "399.,99" feet when i hit rth it should have descended to 200 feet but didnt

b14ckyy, 2018-10-31:
ATM I have no inav plane I can test in that windy conditions here but can someone test the "nav_rth_home_altitude" value in CLI? set it to the same value like "nav_rth_altitude" and then it should go to this height after entering the circle. Possibly this is WAD since 2.0.

digitalentity, 2018-10-31:
@igi1 @b14ckyy @wx4cb please post `dump`-s of your config.

b14ckyy, 2018-10-31:
[Main 3 INAVMATEKF405SE 2.0.0 Aug 20 2018  192234 (dbdd1656a).txt](https://github.com/iNavFlight/inav/files/2535301/Main.3.INAVMATEKF405SE.2.0.0.Aug.20.2018.192234.dbdd1656a.txt)

wx4cb, 2018-10-31:
Im not home right now, but ill post when im home later tonight

digitalentity, 2018-10-31:
@bnn1044 to confirm the issue. You have the folling config:
```
set nav_rth_alt_mode = AT_LEAST
set nav_rth_allow_landing = NEVER
set nav_rth_altitude = 5000
set nav_rth_home_altitude = 3000
```

When the plain reaches home it keeps at 50m or above instead of descending to 30m and staying there. Is that correct?

igi1, 2018-10-31:
since I first reported the issue I have done some more testing. 

Here is the config that preoduced the descibed issue in the first place:

set nav_rth_alt_mode = FIXED
set nav_rth_allow_landing = NEVER
set nav_rth_altitude = 7000
set nav_rth_home_altitude = 0

I than put:
set nav_rth_home_altitude =7000
hoping that this would make the plane descend further when hitting the loiter pattern higher than fixed rth altitude. But strangely enough with this setup switching to RTH did not work altogether anymore , meaning plane stayed in the flight mode it had before switching.
I then lowered the "nav_rth_home_altitude" setting to 6000  and I think that did the trick. I got the impression, that now the plane kept on descending to the specified 60 meters after entering the loiter pattern at higher altitude.... I could not definitely confirm that though due to bad weather conditions at present here (Germany) 

Conclusion: "set nav_rth_altitude" value must never be same as "set nav_rth_home_altitude".

b14ckyy, 2018-10-31:
On Saturday the wheather could become better so I could get the chance to test this with the Bush-Plane I have. At least in my part of Germany.

wx4cb, 2018-11-02:
here's my MT dump. I haven't flown it since the other day so should be pretty accurate. Wanted to fly it today, but was really windy

https://pastebin.com/MpR7Gu3N

wx4cb, 2018-11-02:
here's my S1100 Dump relative to the day before last when i trimmed it out etc.

https://pastebin.com/6qxzdB6s

igi1, 2018-11-03:
Did some testing today in perfect weather condition. I can now confirm that the initial issue is completely solved for me using the following settings:

set nav_rth_alt_mode = FIXED
set nav_rth_allow_landing = NEVER
set nav_rth_altitude = 7000
set nav_rth_home_altitude = 6000

Plane will now keep descending to 60 meters when entering the loiter pattern at highter altitude.
As a side note though:  if "nav_rth_altitude" and "nav_rth_home_altitude" are set to identical values RTH fails to kick in altogether...

wx4cb, 2018-11-03:
interesting. I'll change mine to that and test tomorrow if i can, here's what i have set them all too. the only real change is the alt_mode. your heights were slightly lower than mine at 300 and 250 feet respectively

# get rth
rth_energy_margin = 5
Allowed range: 0 - 100

nav_min_rth_distance = 500
Allowed range: 0 - 5000

nav_rth_climb_first = OFF
Allowed values: OFF, ON

nav_rth_climb_ignore_emerg = OFF
Allowed values: OFF, ON

nav_rth_tail_first = OFF
Allowed values: OFF, ON

nav_rth_allow_landing = NEVER
Allowed values: NEVER, ALWAYS, FS_ONLY

nav_rth_alt_mode = FIXED
Allowed values: CURRENT, EXTRA, FIXED, MAX, AT_LEAST

nav_rth_abort_threshold = 50000
Allowed range: 0 - 65000

nav_rth_altitude = 7000
Allowed range: 0 - 65000

nav_rth_home_altitude = 6000
Allowed range: 0 - 65000

wx4cb, 2018-11-11:
So I tried this the other day - and i will confirm when i get the next chance to fly, but when i killed the transmitter with the aircraft pointing towards me at 400 feet (RTH is 250) the plane nosedived - by that i mean, 90 degrees earthbound, hit 250 and then tried to pull out. 

From memory, set pitch_rate = 36 means maximum 36 degrees in pitch (up or down), the only thing with "pitch" in it referencing failsafe is set failsafe_fw_pitch_angle = 100. but from memory, that means 10 degrees right?

wx4cb, 2018-11-11:
i do need to lower the pitch angle though methinks, as the 30 odd degrees scared the bejeebus outta me lol and of course, the taranis doesn't init the radio link until you deal with all the "switch issues" on power on lol

wx4cb, 2018-11-11:
[https://www.youtube.com/watch?v=vmNkGI3DtIg](url)

closed 2019-01-24 by stale[bot]