iNavFlight/inav #3919: Matek F411-Wing: 2 soft serial ports....please!

Wcallahan01, 2018-10-10:
## Current Behavior
Matek F411 target only allows for 1 soft serial port. This applies to both the Matek F411 mini and the F411-Wing boards...same issue with both boards in Inav. In BF the user can resource map and get 2 soft serial ports. FRsky Sport users really need the extra serial port so they can have GPS, Sbus, Sport & Smart audio connections.

The current target mapping typically requires choosing between Sport and Smart audio when configuring. This sacrifice doesn’t seem necessary if BF allows for mapping of 2 soft serial ports. I’m not an expert, but it seems possible if the 2nd soft serial was mapped to a pinout like they do in BF.

## Desired Behavior
Add target with support for 2 soft serial ports (this is possible using resource mapping in BF already)

## Suggested Solution
Add target with support for 2 soft serial ports (this is possible using resource mapping in BF already.)

## Who does this impact? Who is this for?
All F411 users and the F411-Wing board is becoming very popular lately!

## Additional context

Wcallahan01, 2018-10-11:
> I guess you can make your own firmware with a second softserial like I did on my SpeedyBee AIO FC.

I do have one setup the same way with R9 Fport, but I’ve been having some weird RSSI behavior. So I’m not trusting it too much yet.

There is a lot of frustration out there with just the 3 UARTS available. Check out the Matek F411/405 Facebook group. 

I just happened to build a wing with the F411 mini 20x20 board before the wing version was released and noticed they give resource mapping to get 2 soft serials in BF. Seems like it might be possible in Inav too with this board becoming so popular lately.

I’m not savvy enough to create my own custom firmware...I’m mainly a nuts and bolts mechanical engineer by day. But I was reading the wiki stuff in how to do it. I’d love to take a crack at it but I’m just way out of date on programming like this.

Wes

ReganGTX, 2018-10-11:
Yeah, I guess I can't speak for the RSSI as I haven't maiden it yet. I'm with you on the nuts and bolts. This f411-wing really needs a RSSI pad and one more uart, In my opinion.
I have the F411-wing in the AR Wing 600 and I have the Matek F7 in the AR Wing 900.

Wcallahan01, 2018-10-11:
Haha I’ve got the same model and 411 board setup. I’m wanting to go back to sbus/sport firmware on the R9 but didn’t want to give up the smart audio.

ReganGTX, 2018-10-11:
Nice, share time! https://www.dropbox.com/sh/oy06c7rm6p2uhf2/AABHRXaZ45OUzNpIn0pcpfmha?dl=0

cmenard, 2018-10-12:
I created an iNav 2.0 hex with a second softserial port on the LED pin.  Never used LEDs on any of my planes, so seemed a logical choice.  I use that for Smart Audio now.

Wcallahan01, 2018-10-12:
How do I get s copy of this hex file???? LED pad works for me. 
We

cmenard, 2018-10-14:
The link in the previous comment

Alejandro-pucci, 2018-10-14:
Thank you very much! Just googled "F411-mini two softserials" and this popped up.

P-I-Engineer, 2018-10-30:
Thanks for making softserial 2 for this fc. Hoping to test fly soon.

Wcallahan01, 2018-10-31:
Is this version going to be added as a supported target in future releases?
Thanks

cmenard, 2018-11-03:
It will be available in iNav 2.1.  Until then here is a link for a 2.01 build that has the recent GPS fixes in it.

https://drive.google.com/file/d/1MKkmMg1P8BFyMYkoHgo4U1JT4sI-__zO/view?usp=sharing

Wcallahan01, 2018-11-03:
Awesome!!! Thank you Sir!

Daniel-1276, 2018-11-04:
@cmenard would this may also be a solution path to map analog RSSI to the LED pad ? I am lost in my build to get an RSSI indication using a L9R long range receiver from FrSky.

P-I-Engineer, 2018-11-05:
looks like softserial1 is using the 3rd adc channel (pin PA0).  Using it as a adc is commented out. You'd need to do a custom build for rssi to work.

Daniel-1276, 2018-11-06:
@P-I-Engineer many thanks for the hint !! So i have quickly checked the MATEKF411 target about the pin assignment. Taking your comment into account I would see the following changes in **[target.h](https://github.com/iNavFlight/inav/blob/master/src/main/target/MATEKF411/target.h)**:

**-> comment out SOFTSERIAL section **
#define USE_SOFTSERIAL1         //Frsky SmartPort on rssi pad
#define SOFTSERIAL_1_TX_PIN      PA0
#define SOFTSERIAL_1_RX_PIN      PA0

**-> uncomment** 
//#define ADC_CHANNEL_3_PIN           PA0
//#define RSSI_ADC_CHANNEL            ADC_CHN_3

OK, so now the github-noob question, how do I compile my newly created "MATEKF411_RSSI" target with latest stable release of INAV ? Checked [build instructions](https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows.md) But unsure if master branch is only containing latest "stable" changes or if I have to somehow choose branches for released versions. Any help highly appreciated :)

P-I-Engineer, 2018-11-09:
just tested @cmenard 2.01 firmware, so far so good. I have smart port running on the led pad
thanks

shellixyz, 2018-11-09:
@Daniel-1276 A new target will be available in version 2.1: MATEKF411_RSSI with RSSI input on ST1 pad and softserial port on the LED pad.

Please test with this build of 2.0.1: 
[inav_2.0.1_MATEKF411_RSSI.hex.zip](https://github.com/iNavFlight/inav/files/2566214/inav_2.0.1_MATEKF411_RSSI.hex.zip)

Daniel-1276, 2018-11-09:
@shellixyz many thanks, that's very kind of you ! Will test it by end of next week and provide feedback.

shellixyz, 2018-11-13:
New targets making analog RSSI and 2 softserial ports available on MATEKF411 will be available in the next release, closing.

closed 2018-11-13 by shellixyz

noagendafpv, 2019-01-23:
I loaded the 2 soft serial FW to my F411-W last night and tried to setup the ports for SmartAudio and SmartPort. The ports tab shows UART1, UART2, SoftSerial2 and UART1 again at the bottom. I assumed the second UART1 is the Soft Serial 1, and set SS2 to S.Port and Uart1(Bottom of list) to SmartAudio. Saved and rebooted and then the FC would not communicate with the iNav configurator. I tried restarting the config, tried a different USB cable, tried restarting my laptop and none of that worked. I then reloaded the stable 2.0.1 FW and the FC was back to functioning properly.

Additionally is there any reason that the S7 pin out cannot be used for the second Soft Serial pin? I use the LEDs on 2 of my 3 planes running the F411-wing FCs so I would prefer to not have to give up that ability.

Thanks for the work that has been put into this fork!

Daniel-1276, 2019-01-23:
@shellixyz Just for reference, proposed RSSI firmware worked flawlessly, many thanks again !

teckel12, 2019-01-23:
Also, if you want 2 soft serial ports AND RSSI you can do it via telemetry back to the transmitter and back to the FC on an open channel. I do this for all my builds as it's just easier with some receivers not outputting an RSSI signal the FC supports. No need for an extra wire either.

Daniel-1276, 2019-01-23:
Yes, did this in the past also with other receivers, but not possible for the frsky L9R. It does not do telemetry, nor injecting RSSI information into one of the SBUS channels like the XM+ for example. If you have telemetry, but no injection of RSSI information into a channel, you can do a trick with opentx and use the RSSI information in the TX and report back on any channel you like. Quite neat little trick :)

teckel12, 2019-01-23:
@Daniel-1276 Ah! Didn't catch you were using the L9R.

Tduru, 2019-01-30:
so does the Matekf411_RSSI file allow the user to make the led port for the analog rssi while still using STI port for PPM? If so how to do i go about setting it up

Christianmix, 2019-02-11:
> Novos alvos tornando o RSSI analógico e 2 portas softserial disponíveis no MATEKF411 estarão disponíveis na próxima versão, fechando.

Hello, I downloaded the matek f411 rssi 2.1.0 rc3 version of inav, so that RSSI works I need to pick something or just connect the wire that comes out of the L9R direct in the ST1 and will it work?

Daniel-1276, 2019-02-11:
@Christianmix that's exactly what I did with the custom RSSI version 2.0.1 from shellixyz. Worked great so far on the bench ... Weather to bad in the northern hemisphere at the moment, but I am optimistic this will work out without any smoothing capacitor or something like that.

shellixyz, 2019-02-11:
@Tduru No you can't use RSSI and PPM

SirHaugen, 2019-03-03:
Hi guys!
Im running Matek F411 Wing in my Mini Talon.
The receiver is a Rangelink UHF (PPM)

Will it be possible for me to use this setup and still get RSSI in my OSD?
The Rangelink receiver has a RSSI pin out.

Daniel-1276, 2019-03-04:
@SirHaugen ShellyXYZ already commented about your question. ST1 pad can rather be RSSI (special firmware) or PPM if you disable softserial in your flight controller software. Maybe you could build some custom firmware to reuse the LED pad for RSSI, but that is unconfirmed to work.

oxy55, 2019-04-01:
Hi!
Is there any way to use sbus with an software serial on the f411 wing? It works fine on an uart, software serial however doesn't seem to work. I tried both the normal f411 and the 2 softserials firmware.

Tduru, 2019-04-01:
Have you enabled the Soft serial ports in the configuration tab

oxy55, 2019-04-01:
thanks for your help. For sure, as well as activated serial rx in the ports tab...

P-I-Engineer, 2019-04-01:
you may need to invert it in the cli.

shellixyz, 2019-04-01:
@oxy55 Please do not use a softserial port for S.Bus input. It is a very bad idea. The softserial port can't keep up with the 100kpbs of S.Bus you will have issues.

oxy55, 2019-04-02:
@shellixyz thanks for the hint. Thought it should work as I did it this way in the early days of Betaflight. Is there any other way to use both uarts and sbus separately on the f411 wing?

shellixyz, 2019-04-02:
@oxy55 Unfortunately there is not. But maybe you can use the softserial for one of the other things you want to connect to the HW ones ?

P-I-Engineer, 2019-04-02:
you have 4 uarts with the sftsrl2 firmware.

I'm using

Uart 1: sbus
Uart: gps
soft serial(ST1): smart audio
soft serial (LED pad): s.port

oxy55, 2019-04-02:
I need uart 1 for gps and uart 2 for serial passthrough with TBS Crossfire. For now Im using ppm, however having sbus or even crsf would have been nice.

shellixyz, 2019-04-02:
What do you need a third UART for ? Not sure I would recommend but you could use the GPS on softserial it should work fine.

P-I-Engineer, 2019-04-02:
use uart 1 for crossfire
TX1 and RX1

use uart 2 for gps
TX2 and RX2

oxy55, 2019-04-02:
@P-I-Engineer this way I can't use the serial modem function of the crossfire, I would need a third uart for that

shellixyz, 2019-04-02:
Oh ok I get it now. What speed does the transparent serial of CF works at ?

tasi0000, 2019-04-08:
Now with the crazydude bluetooth telemetry stuff, there's a need to send the GPS rx/tx on softserial. the GPS can work on lower speeds right? Was thinking:
uart 1 - sbus
uart 2 - smart port (this one on a regular uart for the 57600 speed required for the crazydude stuff to work)
softserial 1 tx/rx - GPS

How does that sound and is it possible?

wx4cb, 2019-04-08:
i would suggest putting the gps on the hardware uart and the telemetry on
the softserial

tasi0000, 2019-04-08:
@wx4cb yes I could but then I couldn't use the crazydude Bluetooth telemetry push due to the speed differences.
The logic is that since GPS can work at lower speed and the Bluetooth implementation not, to put the s.port on a uart and GPS to soft serial. Does it make sense?

Piotr-PB, 2019-10-26:
Hello,
I loaded F411 RSSI firmware and connected RSSI input to ST1. What do I set in iNAV configurator to make it work and display in OSD? Now it displays 0 which is not correct as receiver it getting signal.

alexvazquez, 2020-10-20:
Hi I have my Matek F411 with the following setup. UART1 for Frsky Fport, UART2 for my GPS and then I have my Caddx Vista that requires a free UART and I guess this can't be a Softserial. Any clue of someone that alread had same issue and solved it by moving maybe GPS to softserial so I can leave a physicial one for my Caddx?

Appreciate any clue, help or guidance.

wx4cb, 2020-10-20:
A caddx vista should be able to go on soft serial if it works like smart audio. your issue is going to be where it requires both a transmit AND receive line. i don't know anything about the vista, but i would assume it does? If it only needs a transmit line then yes it can work with soft serial for sure.

Just my 2p

MartinHugh, 2022-07-19:
I have installed this 2 softserial target for iNav 5. Many thanks.

I would like to put GPS on softserial 1
I infer that I connect the GPS RX to ST1
But what do I connect the GPS TX to ?

Thanks.