iNavFlight/inav #3921: Timer 1 Channel 3 on PB1

KingKone, 2018-10-11:
Hello inav team,

i have a quad FC (omnibusF4 target) lie around on my bench.
It has only 4 PWM ports and i want to use it for a airplane.
So i thougt maybe i can change the code to use the timer for PWM 6 on PWM 2 port. 
Acording to the STM datasheet Timer1 Channel 3 should work with pin PB1.

So i changed target.h line 39:
`    
//{ TIM3,  IO_TAG(PB1),  TIM_Channel_4, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM3, TIM_USE_MC_MOTOR                    | TIM_USE_FW_MOTOR }, // MOTOR_2
`
`
{ TIM1,  IO_TAG(PB1),  TIM_Channel_3, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM1, TIM_USE_MC_MOTOR                    | TIM_USE_FW_SERVO }, // MOTOR_6 -> MOTOR_2
`
And commented the PWM 6 line.

As i messured the output:
PWM 1: 400 Hz
PWM 2: 0 Hz
PWM 3+4: 50 Hz

Is there an additional step to get the output on PWM 2?
Or do i miss something here?

Mixer is also set 
S1 -> Motor 1
S2 -> Servo 2
S3 -> Servo 3
S4 -> Servo 4

Also thx for this great software!
Daniel

shellixyz, 2018-10-11:
PWM on PB1 with timer 1 can only work with CH3's complementary output you need to change the line to:

`{ TIM1, IO_TAG(PB1), TIM_Channel_3, 1 | TIMER_OUTPUT_N_CHANNEL, IOCFG_AF_PP_PD, GPIO_AF_TIM1, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO }, // MOTOR_6 -> MOTOR_2`

KingKone, 2018-10-11:
As i messured the after your fix:
PWM 1: 400 Hz
PWM 2+3+4: 50 Hz

Thank you very much!

closed 2018-11-13 by shellixyz