iNavFlight/inav #3923: Inav and Matek F411 Wing

rwilabee, 2018-10-13:
I wanted to provide a bit more info to a question I asked before.

Can the Inav Matek F411 Wing with a GPS connected be used Stand Alone as just OSD without motors or a receiver.

For instance I have a Hex Copter with a DJI Controller.  Their OSD is in Metric only and the Inav has the ability to choose Imperial Units.  The main reason I'm wondering is I can get speed to read out but not Altitude when I fly the Hex copter around.  I have all sensors showing as blue on the Inav Screen.  The GPS is showing data coming out of it on the Inav GPS tab.

Thanks

Rich

For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)
* [IRC channel](irc://irc.freenode.net/#inavflight)
* [Slack channel](https://publicslack.com/slacks/inavflight/invites/new)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)

ShikOfTheRa, 2018-10-13:
Might need some sw hacks to the code to get it working. 

Which DJI? Naza?

ShikOfTheRa, 2018-10-13:
I think altitude is reset to zero when disarmed.

Try cli set  inav_reset_altitude NEVER

If you have dji naza you can use several other osd's that can tap into the data from the gps that give a lot of naza info .

rwilabee, 2018-10-13:
It is telling me invalid name

I typed in set inav_reset_altitude NEVER

ShikOfTheRa, 2018-10-13:
ok try:
set inav_reset_altitude = NEVER
also:
set inav_reset_home = NEVER

rwilabee, 2018-10-13:
> The first command did it.  Should I also do your second suggestion.

That worked great.   I did both commands.

Do you know which OSD would work with the DJI o get some of their data that
would give me Imperial units instead of metric.

Thanks again.

wx4cb, 2018-10-25:
theoretically you  can, with one caveat. the osd and the video signal would have to come via the matek board and not the DJI FC. therefore you would lose any and all information that their osd would provide.

i've done this using inav for the voltage readout and gps coordinates on a nano FC but flying it manually.

never tried running it in conjunction with another FC but as long as the only thing you're connecting wouold be power and the video signal i guess it would work.

closed 2019-01-08 by stale[bot]