iNavFlight/inav #3925: Question: will ANGEL mode auto enable after CRUISE mode is activated?

davidngrc, 2018-10-14:
in v2.0 , will ANGEL mode auto enable after CRUISE mode is activated?

I am confused the description here 
https://github.com/iNavFlight/inav/wiki/Navigation-modes#navcruise
In both modes ANGLE mode is active and the plane will stay leveled.

is that mean?
In both modes (2D CRUISE or 3D CRUISE), if ANGLE mode is active, then the plane will stay leveled.
or
In both modes (2D CRUISE or 3D CRUISE), ANGLE mode is auto actived, and the plane stay level.

sorry for my english...

giacomo892, 2018-10-15:
If you enable 2D CRUISE or 3D CRUISE, ANGLE mode is implicit and activated so the craft will stay level.

Please close. This is not a support channel.

davidngrc, 2018-10-15:
thank you.
hope someone will make the wiki more clear.

closed 2018-10-15 by davidngrc

giacomo892, 2018-10-15:
Improved.