iNavFlight/inav #3927: SPRACING F3 Boot problem

HAMZAALI1962, 2018-10-15:
I am facing a problem with a new SPRACING F3 DELUXE with OSD. The board does not boot using INAV.
The problem occurred after upgrading the firmware for the board to the latest version. Also now the red LED flashes two long blinks followed by three or four fast blinks.  I have re-flashed the board with different versions of the inav the firmware but with no luck. I have also tried manual baud rate etc....
Appreciate any help.

For immediate help, just ask your question on one of the following platforms:

* [Telegram channel](https://t.me/INAVFlight)
* [Facebook group](https://www.facebook.com/groups/INAVOfficial)
* [RC Groups thread](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-%28navigation-rewrite%29-project)
* [IRC channel](irc://irc.freenode.net/#inavflight)
* [Slack channel](https://publicslack.com/slacks/inavflight/invites/new)

You can also read public documentations or watch video tutorials:

* [Official documentation](https://github.com/iNavFlight/inav/tree/master/docs)
* [Official Wiki](https://github.com/iNavFlight/inav/wiki)
* [Video series by Painless360](https://www.youtube.com/playlist?list=PLYsWjANuAm4qdXEGFSeUhOZ10-H8YTSnH)
* [Video series by Paweł Spychalski](https://www.youtube.com/playlist?list=PLOUQ8o2_nCLloACrA6f1_daCjhqY2x0fB)

shellixyz, 2018-10-15:
If the LED is flashing it means it is booting up

HAMZAALI1962, 2018-10-16:
Thank you for your feedback. It is a boot up issue. Question is how can I fix it?

shellixyz, 2018-11-13:
Did you manage to make your board working ? The blinking pattern you described means that the flight controller is failing to initialize the acc/gyro chip. It often means you used the wrong target or that the acc/gyro chip is defective.

HAMZAALI1962, 2018-11-13:
Thank you for following up. Much appreciated. 

Unfortunately I still have not resolved this problem. In fact I have two flight controllers with the same issue and what is surprising to me is that both were working until I decided to upgrade the firmware and then they stopped working and both giving the same error blinking pattern. I am sure the problem is not a physical defect and it must be a software issue.

Can you please explain in more detail what you mean by wrong target? Do you mean wrong version of software?

shellixyz, 2018-11-14:
I mean that you may have selected the wrong type of board when flashing. There is more than one version of the  SPRACING F3 board:

- SPRACINGF3
- SPRACINGF3EVO
- SPRACINGF3MINI
- SPRACINGF3NEO

Maybe your boards are not what you think they are. Before upgrading you should connect to it and check what type of target (type of firmware for your board) is flashed onto it by looking into the connection log at the top of the configurator window. See this example for an Omnibus F3 board:

[image]

HAMZAALI1962, 2018-11-17:
Unfortunately the board is upgraded now, however I am sure the board is SPRACINGF3. 
If there is anything else I could do, then please let me know.

On a different note, I am trying to set-up a speed sensor "Flysky CPD02" using INAV configurator. I have connected the sensor to my received Flysky  FSIA-10B. However I can not get the sensor to activate in INAV. The red led on the sensor is blinking.  I have searched on the net but I could not find any useful info on this sensor and how to set it up.  Can you help?

shellixyz, 2018-11-18:
You could try flashing another target even though it is not recommended to flash another target than the one your board was made to work with. But if it is not working already with the target you flashed it must be the wrong target. Did you try to flash Betaflight to see if it makes a difference ? Also you could track where the acc/gyro chip is connected to the MCU with a multimeter. It could help to identify the right target.

About the FS-CPD02 iNav doesn't support RPM sensors and sensors made to be plugged on the receiver don't transmit data back to the FC. The FS-CPD02 will only allow you to see the RPM of the motor on your remote control.

HAMZAALI1962, 2018-11-19:
Thank you for your response.  I have flashed the board with Betaflight and clean flight but still did not work. As I mentioned  earlier, the board was working until I upgraded the software and then it became problematic. The board is SPRACING F3, should I try and flash it with another version  of the board example F3 EVO or F3 NEO? or will this mess up the board more because currently I can flash the board but it does not connect to INAV or any other software? 

Regarding the CPD02 sensor, I have connected it to my receiver but I can not see the RPM in my transmitter FS I10.  I try to add the sensor in FS I10 but it does not detect the sensor.  Do I need to do any setup in the transmitter?

HAMZAALI1962, 2018-11-19:
Further to my last message, I have managed to successful flash the board in INAV. I flashed it with Omnibus as target and it worked. So it seems that the supplier had the board mixed up!  So thank you for your help. Very much appreciated. 

I look forward to your input on how to set up the sensor in my transmitter.

wx4cb, 2018-11-19:
Just a note, the first thing i do when i get any fc, is load up betaflight
and check the version string

shellixyz, 2018-11-19:
> I have managed to successful flash the board in INAV. I flashed it with Omnibus as target and it worked

Very good, glad that it is now working !

> Regarding the CPD02 sensor, I have connected it to my receiver but I can not see the RPM in my transmitter FS I10. I try to add the sensor in FS I10 but it does not detect the sensor. Do I need to do any setup in the transmitter?

I never tried this sensor but you should be able to see it if connected to your receiver, receiver powered on and bound to your remote control. I don't know about other remotes but with a FS-i6 you should be able to display the RPM value on the screen by choosing the sensors in the menu: System -> RX Setup -> Choose sensors. That's all I can say. If you still can't make it work I would suggest to ask for help on a RC related forum where it is more appropriate like [RCGroups](https://www.rcgroups.com/forums/index.php).

closed 2018-11-19 by shellixyz