iNavFlight/inav #3932: No telemetry w/ Kakute F4V2 + FrSky R9 Slim (Flex Firmware)

fchansa, 2018-10-17:
Setup:
Kakute F4V2
FrSky R9 Slim (Flex Firmware, SBUS+Smartport, 868 MHz, 10 mW)
Last OpenTX  2.2.3 (nightly)
RX connected to UART 3 RX and SP(=UART1) for Telemetry
Telemetry enabled in Configurator, Receiver = Serial RX (SBUS)
Ports: UART1 -> Smartport, UART3 -> SerialRX
Receiver (in the Receiver Tab) works fine, but my Horus X10S discovers NO new Sensors.

WITH EXECTLY THE SAME SETUP IN BETAFLIGHT 3.5.2 TELEMETRY WORKS !

In BF there are 2 parameters for telemetry (inversion & duplex), in iNav only 1 (inversion) ... maybe this is the reason ???

teckel12, 2018-10-17:
@fchansa What are your BF/INAV settings for inversion and duplex?  Does the Kakute F4V2 have a pin that controls inversion?  The F4 doesn't do inversion, it needs a circuit to do the inversion.  Most FC have an inverter for SBUS.  This inversion is turned on/off sometimes with solder pads, sometimes with a pin, and other times can't be controlled at all.  In other words, the inversion setting may be doing nothing.  For example, on my Omnibus F4 Pro inversion for the Rx does nothing as inversion is controlled via hardware.  I can invert SmartPort telemetry, but only on a soft serial port (again, because the F4 can't do inversion on it's own).

fchansa, 2018-10-19:
(working) Betaflight settings:
tlm_inverted = ON
tlm_halfduplex = OFF

iicc1, 2018-10-20:
I have exactly the same problem with Kakute F4 V2 and Frsky R9 Mini with Flex firmare.
Telemetry works fine in Betaflight but not on Inav with exactly the same configuration.

I connect S.Port from R9 Mini to "SP FrSky SmartPort Telemetry(UART1, inverted)"

The settings that works with BF are the following (the default ones):

# get tlm
tlm_inverted = ON
Allowed values: OFF, ON

tlm_halfduplex = OFF
Allowed values: OFF, ON

pid_in_tlm = OFF
Allowed values: OFF, ON

In Inav I select telemetry_inverted on/off with no result.

Thanks.

fchansa, 2018-10-21:
@teckel12 @iicc1 
Problem solved:
Playing with parameters - set smartport_uart_unidir = ON was the solution :-)
(Is it equal to tlm_halfduplex in BF ? Confusinh parameter names, should be renamed to tlm_*** ?)

digitalentity, 2018-10-21:
@fchansa yes I agree, that names are somewhat confusing, however unidir/bidir parameter is only used by smartport hence the prefix. Some refactoring might help though 😄

iicc1, 2018-10-21:
Awesome, it works now!!
Many thanks @fchansa

Sedim, 2018-10-22:
Hi guys, I have the same problem with Horus x10s but your solution above did not work for me:
I get three (3) readings only A2, RSSI, and RxBt. But I get those even when smarttport is NOT physically connected.
------
INAV 2.0.0
telemetry_switch = OFF
telemetry_inverted = ON
smartport_uart_unidir = ON
-------

The default settings in BETAFLIGHT 3.5.2 works fine and I get many sensors that are accurate.
 
Setup is:
- Kakute F4V2
- Horus x10s, OpenTX 2.2.2
- Frsky X4R-SB
- RX connected to UART 3 RX and SP(=UART1) for Telemetry
- Telemetry enabled in Configurator, Receiver = Serial RX (SBUS)
- Ports: UART1 -> Smartport, UART3 -> SerialRX

Any ideas?

fchansa, 2018-10-23:
@Sedim 
Have you tried smartport_uart_unidir = OFF as described here (for the X4R-SB):
#3526 ?

Sedim, 2018-10-23:
@fchansa 
I tried all combinations for:
smartport_uart_unidir = ON, OFF,ON,OFF
telemetry_inverted = ON,ON,OFF,OFF

Still no smartport telemetry. 
Again this works fine in  BETAFLIGHT 3.5.2 .

teckel12, 2018-10-23:
@Sedim `telemetry_inverted = ON` doesn't do anything on a F4 FC unless your using a soft serial connection. You'll either need to use soft serial and invert, or not invert and use a not-inverted hack on your Rx.

Sedim, 2018-10-24:
Thank you for your email Tim. I am new to this. The Kakute F4 v2 has UART 1 already inverted so is UART 3 on this F4 FC, its built-in. In fact UART 1 is labeled SP for Smartport and the Kakute instructions indicated that it is inverted. The Smartport from X4R-SB is connected directly to UART #1. Also this setup works fine when I use BETAFLIGHT 3.5.2. So do I still need a not inverted-hack on the Rx with INAV?

teckel12, 2018-10-25:
@Sedim SBUS is working fine, there's a hardware inverter on the FC taking care of things.

Telemetry, however, is also inverted with SmartPort. Unless there's a secondary inverter on your FC for SmartPort it's not going to work. Not sure how it would work on Betaflight, unless your FC has another UART designed for SmartPort telemetry with another inverter on it (which is possible I guess).

Sedim, 2018-10-25:
@teckel12 ". . . unless your FC has another UART designed for SmartPort telemetry with another inverter on it (which is possible I guess)."

Yes, Uart 1 pin is labeled "SP" for "SmartPort" on the FC and is indicated on the Kaukute F4 v2 as an inverted port. The docs for this specific FC recommend connecting SmartPort to this pin and selecting "SmartPort" when configuring this port in Betaflight. **That's why it works as it's supposed to when flashed with Betaflight without  changing anything on the hardware. But still does not work with INAV.**
[image]

teckel12, 2018-10-25:
@Sedim While it shouldn't matter, set  `telemetry_inverted = off` in INAV as that would invert via soft serial or with the F7/F3 microcontroller. With the F4 on that UART there's no reason to invert if there's a hardware inverter.

Sedim, 2018-11-10:
Update: I am finally getting all the telemetry data back to my Horus through the smartport from UART 1 (AKA sp port) on the Kakute F4 V2. I used the following to make it work:

smartport_uart_unidir = ON

VadimGromov, 2019-01-20:
Having the same issue with kakute v2. Connected fport on uart1. It sends telemetry to r9mini, but if I install r9slim or slim+ I have the RC but no telemetry. smartport_uart_unidir givs no effect.