iNavFlight/inav #3933: Antenna Tracker function and firmware

Wcallahan01, 2018-10-18:
## Current Behavior
No antenna tracker functionality or firmware

## Desired Behavior
Antenna tracker firmware for F3/4 or even F1 flight controllers and Inav supported telemetry protocals.

## Suggested Solution
Borrow Ardupilot antenna tracking code or similar (ghettotracker, open360) and implement as an Inav firmware option. Simple config in the GUI would be nice too.
The same cheap f3/4/1 boards that already have targets in Inav could be used to control an antenna tracker. There isn’t a good cheap option for antenna tracking that is as easy as Inav is to setup and use....f3 board and gps for $25-30usd and OLED for info screen would be awesome! Just add a servo pan/tilt for the antenna and your done. Having Inav on both air side and ground side would be much simpler interfacing and both would have the same telemetry libraries. 

## Who does this impact? Who is this for?
All fpv users!!!

## Additional context
There are several open source projects with antenna tracking code that could be ported into Inav. This would make a great package for pilots. It would unlock the hardware constraints of all the other options since Inav supports so many different boards.

Wcallahan01, 2018-10-18:
> I'm confused as to what this has to do with iNav?
>
> You can already use ghettostation with a number of iNav provided telemetry options (LTM, Mavlink, MSP) and it works perfectly. Cheap and easy already.

Like alot of Inav users I’m not a code guy. I can understand the parameters and logic. But writing or compiling code to flash escapes me mainly because of a lack of spare time. Im trying to learn and dig into but I’m a mechanical engineer for a reason. I get the parameters, logic and controls part. But the code and compiling isn’t easy for many users.

The Inav Interface and method is awesome! And the number of target boards is extensive. So the benefit is that code dumb guys like me could load a tracking program on a large variety of different boards, flash and config easily.
Also, doing this opens up more development opportunities for antenna tracking because the FC boards have more sensors and loads more horse power than the Arduino.

An Omni F3/4 board would make for a much cleaner tracker build! 5v bec for servos, OSD display chip, accelerometer, gyro, baro, gps and  compass connection are all on the board already. And Inav already has a good library of telemetry protocols and it’s simple to switch between them in the ports tab. Most tracker projects require loading different firmware for different telemetry protocols.

I wish I had the skill to grab another projects code and create a custom firmware to load on an OmniF3/4 or similar board. But I just don’t; or haven’t made it that far yet. And I think the Inav experience has just spoiled me by making it pretty darn easy to grab a FC flash, wire up, and config. 

Please consider it and understand that most of us are either not that smart or we just don’t have enough spare time to learn all the magic under the hood.

rb1205, 2018-10-22:
Sorry but I don't really think anyone is going to add unrelated code to a project so that you won't have to learn how to use something specifically made for that exact purpose.

Fameing, 2018-10-22:
For example project 
https://github.com/raul-ortega/amv-open360tracker run on  NAZE32

Wcallahan01, 2018-10-22:
I understand the idea you all are sharing and this is not an attempt to subvert learning how to work out the tracker myself...I’m already 3/4 done with my build and setup. 

Ive become the local Inav support for my flying group when others have issues and are learning to setup. Inav has gotten very popular and the fact is many users don’t fully understand the finer details or control logic. I have a lot of time invested in my tracker build right now because I actually try to read the code and understand the logic. 

My thought was that leveraging the Inav platform, libraries and user friendly GUI would be a great opportunity to add tracker functionality. And I think it would benefit a lot of users.

Ardupilot went this route of including a tracker functionality based on its development requests and lifecycle. This likely occurred because the user base ultimately wanted it. 

One of the biggest benefits I see is that Inav has targets supporting tons of different flight controllers. The only one I saw used was a naze32 board which has been discontinued and difficult to find for a new users. 

It’s just my suggestion; and if I was better at coding (I’m a hack at best); I’d find some way to contribute to make it happen.

teckel12, 2018-10-23:
@Wcallahan01 Maybe you could describe what "tracker" even is. I'm totally lost as to what your trying to accomplish. I use GPS with INAV to track my models. How is this different?

Wcallahan01, 2018-10-23:
Yes, it’s an fpv receiver antenna tracker. It’s a separate device that receives telemetry data on the ground and uses servos or steeper motors to position the fpv receiver antenna at the model in the air. 

It’s very useful and necessary when using high gain antennas that have narrow beam widths.

Wes

Fameing, 2018-10-24:
@DzikuVx  But it is better to have this functionality inside the INAV. For full operation, two servos with 360 turns are enough.

DzikuVx, 2018-10-24:
It's not really a "feature" more like a completely new product

Wcallahan01, 2018-10-24:
Would anyone be offended if I did a poll on the Inav Facebook group to see if there is a great deal of user interest in adding the antenna tracker programming and config to Inav? 

It’s just a thought, I tend to be a little stubborn so feel free to blatantly tell me to stop asking 😜

Wes

rb1205, 2018-10-24:
Inav SENDS positional data, while a tracker RECEIVES positional data. It's a completely different scope, the required code doesn't have anything in common with anything else save for positional sensors drivers and servo output. Even if they'd add some tracking function to orient a directional antenna toward the base position, which could make sense ON THE MODEL, it wouldn't still be able to receive positional data to work as a ground base tracker.

This is similar to you asking "Hey, I love coffee. I noticed that Inav heats up the motors, so would you mind adding some code to heat my coffee as well?"

Wcallahan01, 2018-10-24:
I am not very appreciative of your response as it’s worded. Actually, it’s pretty rude. I’m not going to engage in a elaborate rebuttals, calling out the errors in your commentary or suggesting that you didn’t even take the time to read the whole conversation. Since you said “they’d”, I can only assume your not even a contributor. But since you brought it up, the antenna positioning on the craft might be pretty cool for smaller 5.8 antennas....but might be impractical. 

Simply stating that the dev group has no interest in adding that functionality would have been sufficient. As I previously stated, it was just a suggestion and I was open to being politely told to stop asking. 

Wes

digitalentity, 2018-10-24:
@Wcallahan01 antenna orientation device is out of scope of INAV, so, yes, developers are very likely to be completely not interested in adding this functionality **to INAV**. However, if somebody would use INAV codebase, create a fork and make it an antenna tracking platform - that would definitely be welcomed by the community.

Wcallahan01, 2018-10-25:
Ok, thank you for the gentlemanly reply/resolution. 
Wes

raul-ortega, 2018-11-24:
Hi all... what Wcallahan01 is suggesting is exactly what I did with a old version of cleanflight. I used it as a base code to create a firmware (u360gts) to controll an antenna tracker system reusing old flight controllers stored in a box. And some months ago I had the crazy idea of do it with iNav, something similar as arduipilot does...exactly the same thing that Wcallahan01 is trying to explain. I'm very surprised reading somebody with the same ide I had then...

I would very pleasant to to contribute working on this Idea, but life is too short and I have many things in mind, some projects that I'm planning to start with, and many responsabilities ... I have been alone developing u360gts since I started it on 2015, no contributions from other developers, It would be too much work only for a person.  If other developers join the idea with interest then I could think about the possibility of migrating to iNav.

datamann, 2018-11-24:
I'm not a professional developer but I will gladly help if I can.
How do you start a project like this?
What would be the first thing to do?
Create a feature list maybe?

padcom, 2019-07-08:
Guys, I've been looking around at possible solutions (dyi and off the shelf) and it struck me that if iNav would be somehow capable of producing an audio feed that sends telemetry data over the audio channel (basically a waveform generator with encoded digital data) it would be easy to capture that data then on the standalone tracker and use that to position the antenna.

I know it is a lot to ask, but maybe there would be any kind of chance to see it happening in the future?

I would like to not add hardware to the plane and it seems that a single digital output should be more than capable of generating a low-frequency audio with the encoded information. I mean if we're already doing software serial what would be so strange in generating an audio feed with telemetry?

wx4cb, 2019-07-08:
@padcom there is already a couple of solutions to this, you can:

1. run mission planner with a mavlink stream back from the aircraft and use a pololu maestro to control the tracker

2. Use the u360gts open source tracker to do the same thing with a mavlink stream. there's a number of guys in the LRH facebook group that already do this

3. MWP-Tools by @stronnag has just implemented a api of sorts and i'm working on an antenna tracker module for this utilising a "plugin" and an arduino. it's still in the dev stages, but so far im getting the data. my main problem right now is a crossfire issue i'm trying to fix - not sure if it's the module or the CF firmware, but i'm working with TBS on it directly. Once that is resolved it shouldn't be too long.