iNavFlight/inav #3935: Low freq wable

access3424, 2018-10-20:


rb1205, 2018-10-20:
Low freq wobbling **usually** means too high I factor. But we'll need a log to say anything more.

access3424, 2018-10-20:
Well logging doesn’t work because the matekf405wing is especially picky on what sd you put in. So far cant make it work on any sd i used

b14ckyy, 2018-10-20:
Try formatting with https://www.sdcard.org/downloads/formatter_4/

Do not use SD bigger than 8GB

DzikuVx, 2018-11-11:
low freq wobble is either too much Igains or a gyro LPF is too low (for example 42Hz)

closed 2018-11-11 by DzikuVx

access3424, 2018-11-11:
Whats the difference between the two gyro LPFs? The one in the configuration tab and the one in the PID tunning tab? It sais the one is hardware and the other software but what does this mean for me nd at what values should they be for lets say a 12’ quad.