iNavFlight/inav #3936: Fixed wing, if trimming on transmitter is used flight is impossible

druckgott, 2018-10-20:
## Current Behavior
If I use my trim on my transmitter it's totally impossible to fly a fixed wing any more. The problem is that if you move the stick to fast to the side where you also moved the trim the surface do not any more response only if you do it slow. This is also on manual mode

## Steps to Reproduce

1. Go to manual mode
2.trim the aileron to the ride side
3. Move the aileron stick slow and fast to the right side. Slow it works fast it will not move
4.

## Expected behavior

Always move same speed in manual mode also if the trim was moved a little bit
I see this also as really safety critical. Topic if you move the trim because you do not take care or accedently the fixed wing today was really hard to get back. 
## Suggested solution(s)

Like expected behavior ore maybe it's possible if the signal for trim is different then the stick. Movement itself that the inav moves the lower and middle. And high value for the channel parallel to the right for this example
## Additional context

---

2.0.0

- FC Board name and vendor: matek f405 osd
- INAV version string: I do not have this here at the moment on my Handy, I can post this latet

b14ckyy, 2018-10-20:
You should upload a Blackbox log too, so the guys can check them for any issues. 

Some Questions: Did you fly in Air Mode or Manual Mode? Did you try to switch to manual mode (passthrough)? **You should never trim on the transmitter anyway.** 

See Fixed Wing Guide: 

> Your transmitter should use NO mixing at all (so separate channels for Thr, Ail, Rud, Ele).
> Check that when moving the sticks, the right channels moves in the receiver window. Also everything should be centered at 1500us, and full stick movement should be 1000-2000us. Use sub trim and travel range on your TX to set this up.

> If airplane is not flying leveled when in self leveling mode like Horizon you need to trim your board aligment 
> If airplane flies leveled, do an Servo Autotrim

https://www.rcgroups.com/forums/showpost.php?p=35059861&postcount=6741

https://www.rcgroups.com/forums/showpost.php?p=36039077&postcount=8732

digitalentity, 2018-10-21:
You should never trim on the TX.

druckgott, 2018-10-23:
Yes I read this later but it will be a nice feature if it's possible for a wing.

And if you use it first time it will help if you do not know this and did not read everything maybe a warning in the inav configuration will be also help to do not the same wrong thing what I did.

b14ckyy, 2018-10-23:
The problem here is, that the Flight controller will always "think" that there is a stick input. It cannot detect the difference between trimmed mid points and permanent stick input into one direction. 

If you always fly in Passthrough mode (only using FC for Failsafe reasons) you could trim on the Tx because the FC will pass through the values from transmitter directly to the servos. But in Acro Mode, no trimming will be needed at all because the FC will manage that the plane does not change attitude. Autotrim will just help the FC to know the correct mid point to better control the plane.

druckgott, 2018-10-23:
But also in this mode I'd did not work for me

b14ckyy, 2018-10-23:
this also descripes trimming via FC and not on Tx. Just don't do it in the future ;)