iNavFlight/inav #3937: [REQUEST] Disable FS-RTH during Missions

b14ckyy, 2018-10-20:
## Current Behavior
If Mission is active and Plane/Copter looses contact to Transmitter, Failsafe kicks in with configured behavior.

## Desired Behavior
If a Waypoint Mission is active, RC-Failsave should be disabled. This should be Opt-In. So it would be possible to run fully autonomous UAV Missions, also out of range of your remote. Sure this should be used with precaution. At the Moment it is possible to disable Failsafe completely but that should never be done when in manual control. 

## Suggested Solution
Before Failsafe kicks in, it should check if the NAV_WP Mode is active and valid waypoints available. If mission is at last Waypoint (no matter if RTH, LAND or LOITER) and there is no RC signal, Failsafe should kick in. Best solution would be, if a Return Path RTH is implemented too, if a mission goes behind a hill or so. 

## Who does this impact? Who is this for?
Long Range Wing pilots for full autonomous UAV Flights and out of range of transmitter. Also Drone Pilots that plan missions that will go behind buildings or hills for observation or structure scanning photography.

b14ckyy, 2018-10-21:
#1742

peteoz, 2019-05-22:
please re-open this request and consider implementing it, thanks.

digitalentity, 2019-05-24:
Fixed by #4731

PR #4731 (2019-05-24): Add option to continue mission on FS
Closes #3937. Tested by @CrazyDude1994
closed 2019-05-24 by digitalentity