iNavFlight/inav #3938: Incorrect Vtx power settings

dovcharov, 2018-10-21:
Good day, support.

I have FuriousFPV Stealth LongRange VTX, which has 4 power modes: PIT mode, 25mw, 200mw, 700 mw. It is managable via IRC Tramp protocol.

When I set power via new feature (set larger output power on first arm, i get incorrect settings. Configurator allows to select values from 1 to 5, and in CLI i can select values 0 to 4. 

When I set 0, 1, 2 as settings in CLI, I get 25 mw on arm. When I set 3 - I get 200 mW. When I set 4 - i get 25 mW.
And, since there's no 5 option anymore, I get VTX settings error when I select 5 via configuration tab in Inav configurator. I also cannot set 600 mW via command line.

Please fix this back.

rb1205, 2018-10-22:
I have a IRC TRAMP and I second this: the 600 mW setting, which sets vtx_power to 5, causes the FC to prevent arming due to invalid setting.

paultbg, 2018-11-01:
The same (or very similar) problem on Matek 411 wing w/ inav 2.0 paired with Atlatl HV 2 operated via Smartaudio. The power levels displayed on the OSD do not match the manual. Then on the largest available power (600mw) I can't arm anymore. The biggest power where I can arm is 400mw.

teckel12, 2018-11-01:
See #3564 found via a search for "VTx power"

DzikuVx, 2018-11-11:
fixed in #4008

closed 2018-11-11 by DzikuVx