iNavFlight/inav #3940: Please, Repair BlackBox Fonctionnality on several FC

CcLaFleche, 2018-10-21:
It's important for fine setting in acro mode and GPS/MAG... It never work on matek f722 !

Thanks.

rb1205, 2018-10-22:
I fly a matek f722 ad i've a working logging. I had to try with 3 different SD cards tho, the one that is currently working is a sandisk ultra 16gb A1, like [this ](https://www.google.com/aclk?sa=l&ai=DChcSEwj2puzCuZneAhXXhtUKHQZQDwUYABABGgJ3cw&sig=AOD64_3B5fuDFbodywgouEJ2yyT5eICklw&ctype=5&q=&ved=0ahUKEwjis-fCuZneAhVLxxoKHd_CDiYQwg8IPw&adurl=)one

DzikuVx, 2018-10-22:
I use F722-STD on two of my quads and in both cases logging on SD work. And yes, Sandisk Ultra 16GB is confirmed to work

closed 2018-10-22 by DzikuVx

CcLaFleche, 2018-10-22:
I will try your card...
What is the probleme for the NXT F7 ?

sno1957, 2019-01-07:
NXT F7 will not save. brings up save window and then gets stuck on the progression bar 0bytes moved.  file was created. i have the 2.1.0 hex loaded that readdresses the SPI for the OSD fix.