iNavFlight/inav #3943: set gyro_hardware_lpf = Crash???

JDFPV, 2018-10-22:
I had an issue that crashed my plane... I followed some of Pawels video on how to setup a wing with inav.... In this video he said to set the set gyro_hardware_lpf = 256hz (just because it's the best way to set it) I was worried about changing the settings considering he did not explain what it does... And I could not find any info  about what it does. But I put a little trust in what he said cause I know Pawel works with INAV... Boy was I wrong! My wing flew okay prior the changes... After i changed them I tested RTH and the plane went nuts causing a crash! So my question today is... What does the set gyro_hardware_lpf and how does it affect the plane other than causing a crash if it is turned up to high on a wing?

giacomo892, 2018-10-23:
Hello. Never blindly apply any setting!
Please post your configuration (diff) and blackbox logs.

DzikuVx, 2018-10-23:
@JDFPV `gyro_hardware_lpf` is descibed in CLI documentation. https://github.com/iNavFlight/inav/blob/master/docs/Cli.md

Too bad you crashed your plane but without blackbox log and CLI dump there is no really a way to determine what caused the problem.

JDFPV, 2018-10-23:
I'm using the matek f411 board. So no blackbox. But I do have osd footage and the dump files I will share with you guys shortly.

DzikuVx, 2018-10-23:
All my UAVs whly with gyro_hardware_lpf so no, it was not a cause of the problem. Combination of events rather. Anyhow, please provide a CLI dump and OSD mayube this will give a clue what really happened

JDFPV, 2018-10-23:
Here's the dump file. 

# version
# INAV/MATEKF411 2.0.0 Aug 20 2018 / 19:23:35 (dbdd1656a)

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000
mmix 1  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0
smix 1 3 1 50 0
smix 2 4 0 -50 0
smix 3 4 1 50 0

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 -100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature MOTOR_STOP
feature SOFTSERIAL
feature GPS
feature TELEMETRY
feature CURRENT_METER
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

[... 384 lines cut ...]
set tpa_breakpoint = 1500
set fw_tpa_time_constant = 0
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 40
set pitch_rate = 15
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430 
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 250
set vbat_warning_cell_voltage = 275
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

teckel12, 2018-10-23:
@JDFPV Looks like something totally different than `gyro_hardware_lpf` being set to 256HZ.  BTW, I set `gyro_hardware_lpf` to 256HZ on a few of my quads.

JDFPV, 2018-10-24:
Problem solved. Unfortunately, i have met 4 other people who made theses changes on there (Fixed Wings With GPS) that blindly followed the suggestions from the (How to setup inav on a flying wing video). One guy said his horizon bar was going nuts, the other 3 seen the same similar problems i had while flying in the air. So i guess all 5 of us have a Combination of events that caused this?? No Way Possible!

closed 2018-10-24 by JDFPV

giacomo892, 2018-10-24:
Would be nice if you tell how you have solved the issue...

DzikuVx, 2018-10-24:
And showed what exactly had happened since you deleted that video.

digitalentity, 2018-10-24:
Generally, `gyro_hardware_lpf=256HZ` shouldn't be the cause of the issue, but in my experience 20HZ is a better setting for a flying wing.