iNavFlight/inav #3946: Emergency Glide - Thrust Loss Detection

b14ckyy, 2018-10-23:
## Current Behavior
If there is a loss of all the thrust on a fixed wing (Lost Porpeller, fried ESC, Fried Motor, disabled ESC because battery voltage drop), the plane does not detect that and tries to continue flying in the actual mode. If this is an automatic mode, this would cause a stall when the plane tries to keep height. 

## Desired Behavior
If in any Auto mode (Mission, Cruise, Loiter, RTH) the FC could continuously check for high frequent vibrations of the motor and Propeller that should be always there on most wings. If there is a massive change in expected vibration behavior, the FC should switch into an emergency RTH glide mode with a fixed pitch down angle to glide as far home as possible instead of stall and fall down uncontrolled. 

In manual Mode it should keep normal behavior but show a warning in OSD that no thrust is generated anymore. 

## Suggested Solution
First the FC could check for expected vibrations. If there is a loss of expected vibrations, the FC could check the Speed of the plane. If it goes lower than a specified speed (close to stall speed) even if the FC rises throttle, INAV should switch into the emergency glide with RTH function to try to get as close to home as possible bevor impact. 

Should be combined with wind speed estimation feature or with airspeed sensor to calculate a correct stall speed value. 

## Who does this impact? Who is this for?
all fixed wing pilots

## Additional context
-

wx4cb, 2018-10-23:
it could also look for a sudden current drop maybe (in the case of a blown ESC or motor).

as far as the "result"  it would depend on the altitude ... you wouldn't want it to rth  when the current terrain is higher than what the plane and you are (ie fly into a mountain).

Personally I would prefer maybe just a loiter mode where it slowly descends. - obviously configurable by the configurator :D

b14ckyy, 2018-10-23:
The most obvious thing I wasn't thinking about… of course to check the current consumption relative to throttle is the easiest thing to detect a fried ESC/Motor or even a lost propeller. Because a Motor without load will also significantly take much less current....

FPVZaphod, 2018-10-28:
In half throttle and a very steep dive the current consumption will be also very low, because the prop is just windmilling in the air. I think this feature is not so easy to implement in a reliable way. 
Maye it's required to just use it's result in level or near level flight, and in a speed range where there is a risk of stall. If you then raise the trottle, and there is no speed gain (and / or an unusal low current), then there must be a fault in the drive train. 
Simple way would be a variable where you enter the minimum throttle for reliable level flight. This can be even same like cruise throttle. If you don't maintain minimum safe level flight speed with this throttle value, the propulsion system can be declared as faulty.

wx4cb, 2018-10-28:
The other way would be to use the telemetry from blheli32 escs as u can get
that info from them directly

FPVZaphod, 2018-11-01:
BLHeli has a "Prop left the Plane" detection? I was not aware of that...

danarrib, 2018-12-13:
I like the idea of having a warning message blinking on OSD informing the pilot of problems with motor.

To keep things simple, I suggest the creation of only two new settings:

- throttle_warning_threshold (=0 means this feature is disabled)
- throttle_warning_min_amps (in deciamps)

Assuming this values are 20 and 10 respectively (20% throttle and 1 Amp):

So, if throttle is > 20%, and current draw is below 1 Amp, then obviously something is wrong and motor is not generating any thrust.
Inav then can blink a message like "POSSIBLE MOTOR PROBLEM".

This can also be used on the future for an automatic flight mode like this one suggested on this issue. But for now, It would be nice to just have this simple warning.

closed 2024-02-02 by DzikuVx