iNavFlight/inav #3947: PPM receiver does not work after Refactor TIMER/DMA framework (#3833)

alexeystn, 2018-10-23:
I tried to use the latest version from Development branch and found that PPM receiver stopped working after #3833 pull request (78bc53f commit). All channel values on the Receiver tab are zeros.
Previous version #3868 (3b1b1bf commit) is fine.

Hardware:
Matek F405 Wing (MATEKF405SE target)
FrSky D8R-II receiver with PPM output connected to SBus input of FC.

Dump:
https://pastebin.com/eDnPW6aY
Equal for both versions. Almost no changes to default settings after erasing and flashing. Only changed RX type to PPM.

Not sure if I am experienced enough to fix the problem by myself, but I'm ready to do any tests on my hardware to help you.

digitalentity, 2018-10-24:
Noted, will troubleshoot

digitalentity, 2018-11-18:
F7 fixed in #4021
F4 can't confirm the bug - works fine.

alexeystn, 2018-11-26:
Checked two FCs:
DYS Omnibus F4 Pro (DYSF4PRO) - PPM works fine
Matek F405 Wing (MATEKF405SE) - problem exists

closed 2019-01-05 by digitalentity