iNavFlight/inav #3951: Smart Audio not working over Soft Serial on F411 Wing FC with Cricket Pro V2.1 Vtx iNAV 2.0

Nihilistic, 2018-10-24:
## Current Behavior

## Steps to Reproduce

1.Soldered white SA wire from Cricket to ST1 pad of F411
2.Enabled Soft Serial in Config tab of iNAV
3.Enabled TBS Smart Audio on Soft Serial in Ports tab of iNAV
4. Attempted to configure Vtx settings via OSD stick menu, but all I see are dashes where there should be channel and power information.

## Expected behavior

Should be able to set band and channel and power of the Cricket Pro v2.1 (and any other SA capable Vtx) over soft serial using SA. 
## Suggested solution(s)

Hopefully fix it? hehe...I think this issue may have lain unresolved since December 2017 (different FC though).
## Additional context

# version
# INAV/MATEKF411 2.0.0 Aug 20 2018 / 19:23:35 (dbdd1656a)

# resources

# mixer

mmix reset

mmix 0  1.000  0.000  0.000  0.000

# servo mix
smix reset

smix 0 3 0 50 0
smix 1 3 1 50 0
smix 2 4 0 -50 0
smix 3 4 1 50 0

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 -100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
[... 870 lines cut ...]
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 275
set vbat_warning_cell_voltage = 330
set battery_capacity = 3200
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

# 

---

- FC Board name and vendor:
Matek F411 Wing
- INAV version string:
2.0

shellixyz, 2018-11-13:
> all I see are dashes where there should be channel and power information.

You probably need to update the OSD font

Nihilistic, 2018-11-13:
@Rclay11541 The F411 Wing only has two UARTs. My Crossfire NanoRx is on UART1, and GPS is on UART2. Which is why I want to use Soft Serial for SA.

There might possibly be some ground issue, as you suggested, because I'm using a 12V step up to power the VTX off of the 2S 18650 lithium ion pack. Powering the camera from the 5V and ground pads on the FC.

@shellixyz and @Rclay11541 Everything else on my OSD works and looks normal on my OSD, so I don't think it has anything to do with the font. I loaded the default font the very first step when setting up OSD.

closed 2019-01-27 by stale[bot]

Ozydego, 2019-03-06:
Has anyone had success with this? I can enable soft serial in Betaflight and use the F411-Wing without issue, as well as use a normal UART for smart audio in inav 2.1.0. It won't work with the ST1 pin though.

Nihilistic, 2019-03-06:
@Ozydego I moved the SA wire from the ST1 pin on the F411-Wing to the TX pad on my Crossfire Nano RX, and can now access all the SA features via the luascript on my Taranis. The only thing missing is the auto power switching feature available through the FC. This would be nice, because you could set the VTX to minimum power while on the ground waiting for a GPS lock, or while tinkering on the bench, to avoid overheating. Then upon arming, the FC would kick the VTX up to your preferred max power setting. As it is now, I just have to remember to bump the VTX power up manually before launching. Smart Audio is essential to my build, because the VTX is embedded in the wing of my Nano Goblin, button side down, to maximize air flow over the heat sink side. 

It doesn't look like this issue is going to get any attention unless more people chime in. I'm surprised there haven't been more complaints, because I followed a build video posted by RMRC on the Nano Goblin using the F411-Wing, and he set up SA on the ST1 pin using Soft Serial, but never actually verified that it worked.

jkflying, 2019-05-04:
I'm using the latest development branch and I have SmartAudio 2.1 working on ST1 pin. It didn't work at first, but I then realised that the support for SmartAudio 2.1 was only merged a few weeks ago, ported across from Betaflight, and my VTX runs SmartAudio 2.1.

(Also on a Nano Goblin, fantastic plane!)

Ozydego, 2019-05-07:
@jkflying can you point me to the development branch?

jkflying, 2019-05-08:
https://github.com/iNavFlight/inav/tree/development

Otherwise if you used `git clone` to get the source, you just need to do `git fetch && git checkout origin/development` to get the latest development branch. 

I've attached the `hex`s from the version I built (16faa4399f2d28699f1be9791494c40c1f2c12f6)

[MATEKF411_16faa4399f2d28699f1be9791494c40c1f2c12f6.zip](https://github.com/iNavFlight/inav/files/3156273/MATEKF411_16faa4399f2d28699f1be9791494c40c1f2c12f6.zip)

jasonsmit4, 2019-06-10:
I can confirm this issue still exists on iNav 2.2 RC2.

RCManiac98, 2019-07-27:
> I'm using the latest development branch and I have SmartAudio 2.1 working on ST1 pin. It didn't work at first, but I then realised that the support for SmartAudio 2.1 was only merged a few weeks [...]

Could you please clarify exactly what steps you took to get SmartAudio working on the ST1 pin?

knutigro, 2021-01-17:
I have same issue on 2.6.0 
/ Knut

fpvMarkS, 2024-07-28:
INAV 6.1.1 and still can't get smartaudio working over softserial using st1 pad