iNavFlight/inav #3952: No SBUS signal on F405 wing and L9R receiver

demhat25, 2018-10-25:
I would really appreciate some help here, cause I am stuck for almost a week now. I have a brand new F405 wing and an L9R receiver. For some reason I get no sbus signal from the L9R to the FC. The receiver is linked and working ok from regular channels, and the FC seems to work ok. 
I have followed all the instructions, setting serial RX for UART2, sbus as communication protocol , checked for continuity from the top of the pin to the back of the board, but no resolve. 
Unfortunately i don't have another FC or sbus receiver to cross check....
Any help will be REALLY appreciated at this point. Thanks

teckel12, 2018-10-25:
@demhat25 SBUS is an inverted signal. The F4 itself can't invert a UART signal. So traditionally there's a particular UART on F4 FC where you must connect SBUS that has a physical inverter on the FC. For some FC, you can turn this inverter on or off.

So first of all, make sure your connected to the SBUS UART and if you need to turn on the inverter, do so.

The other possible issue is that with newer FrSky Rx, they have both inverted and non-inverted signals. So you'd need to be sure you're connected to the inverted or uninverted pins on the Rx depending on your situation.

A `dump` from your FC and the physical connections on your Rx and FC would be really helpful to assist more.

demhat25, 2018-10-25:
Thank you for you time, really appreciate it. 
The UART i am using is a dedicated SBUS UART with a built in inverter. 

Here is a recent dumb
# version
# INAV/MATEKF405SE 2.0.0 Aug 20 2018 / 19:22:34 (dbdd1656a)

# resources

# mixer

mmix reset

mmix 0 1.000 0.000 0.000 0.000
mmix 1 1.000 0.000 0.000 0.000

# servo mix
smix reset

smix 0 3 0 50 0
smix 1 3 1 50 0
smix 2 4 0 -50 0
smix 3 4 1 50 0

# servo
servo 0 1000 2000 1500 100
servo 1 1000 2000 1500 100
servo 2 1000 2000 1500 100
servo 3 1000 2000 1500 100
servo 4 1000 2000 1500 100
servo 5 1000 2000 1500 100
servo 6 1000 2000 1500 100
servo 7 1000 2000 1500 100

# feature
feature -THR_VBAT_COMP
feature -VBAT
feature -TX_PROF_SEL
feature -BAT_PROF_AUTOSWITCH
feature -MOTOR_STOP
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature -FW_LAUNCH
feature -TRACE
feature VBAT
feature TX_PROF_SEL
feature MOTOR_STOP
feature SOFTSERIAL
feature TELEMETRY
feature CURRENT_METER
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD
[... 387 lines cut ...]
set rc_expo = 70
set rc_yaw_expo = 20
set roll_rate = 28
set pitch_rate = 14
set yaw_rate = 20
set manual_rc_expo = 70
set manual_rc_yaw_expo = 20
set manual_roll_rate = 100
set manual_pitch_rate = 100
set manual_yaw_rate = 100

# battery_profile
battery_profile 1

set bat_cells = 0
set vbat_cell_detect_voltage = 430
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 330
set vbat_warning_cell_voltage = 350
set battery_capacity = 0
set battery_capacity_warning = 0
set battery_capacity_critical = 0
set battery_capacity_unit = MAH

#

demhat25, 2018-10-25:
Here is a link of a video as well
https://youtu.be/qMYoCPTmbAg

Thanks again, I am kind of desperate at this point...

datamann, 2018-10-25:
Have you tried setting serialrx_inverted = On ?

demhat25, 2018-10-25:
No I haven't , it is supposed to be inverted by default on the f405, but i will.

demhat25, 2018-10-25:
Do I do it through the cli?

datamann, 2018-10-25:
Yes

datamann, 2018-10-25:
set serialrx_inverted = On

wx4cb, 2018-10-25:
and just to make sure, you ARE using this pin on the FC right?

[image]

demhat25, 2018-10-26:
Yes of course, I tried all your kind suggestions, but nothing worked unfortunately. I think there is a problem with my FC and I have asked for a refund. Fortunately Bangood have been very quick and already offered a refund.

DzikuVx, 2018-11-11:
seems to be stale, closing

closed 2018-11-11 by DzikuVx