iNavFlight/inav #3953: with better uart option on anyfc f7 t..this flight controller have lot of potential . then why there is NO OLED DISPLAY feature avaliable. i am using anyfc f7 from inav 1.7 till now to inav 2.0 still now its not there .so ADD OLED DISPLAY FEATURE

Dhananjaya96, 2018-10-25:
## Current Behavior

## Desired Behavior

## Suggested Solution

## Who does this impact? Who is this for?

## Additional context

b14ckyy, 2018-10-28:
1st: You should keep the feature request policy in this git and stop using CAPITAL LETTERS.

2nd: The Anyfc F7 seems to be hard enough to get it working. It doesn't even has a buzzer or an OSD. this FC is made for Betaflight and race quads and not for INAV. So I could understand why the devs have no high prio to find massive workarounds to get extra features to work. Especially because the Use of an OLED Display is even more rare out in the wilds… 

3rd: Do you even have a F7 with baro or without baro?

Dhananjaya96, 2018-10-29:
Without baro

Dhananjaya96, 2018-10-29:
there is a buzzer pin available for osd anyfc provide additional uarts for that . 
again what you will do with 8 uarts on racing quad .
it is highly recombined that before put your comment on anything . first go through documentation .

b14ckyy, 2018-10-29:
I Have: https://quadmeup.com/tag/anyfc-f7/

> Because both original and Banggood clone of AnyFC F7 flight controller and missing buzzer support, its usefulness is somehow limited. Next release of INAV (1.7.2) will finally solve this problem by reusing Motor #9 output as buzzer output. 

And anyway…. I doubt that the devs will answer or consider your Feature request, if it is not in an appropriate format.

DzikuVx, 2018-11-11:
OLED feature is currently enabled on all the targets supported by INAV.

closed 2018-11-11 by DzikuVx