iNavFlight/inav #3954: Matek F722-STD alternative Hex X configuration with Servo on S7.

Frontier314, 2018-10-25:
## Current Behavior
Currently (INAV 2.1 developer branch) it is not possible to assign a Servo (e.g Servo 0) to S7 output pin on MATEK F722-STD target. The issue is explained in detail here: https://github.com/iNavFlight/inav-configurator/issues/588

## Desired Behavior
Ability to set servo on S7 pin on Hex X mixer without disabling Motor on S6 pin.

## Suggested Solution
Either provide an alternate target (e.g MATEKF722-HEXSERVO) with servo assigned to pin S7 or allow to set the desired output pin to use for a servo via the smix command.

## Who does this impact? Who is this for?
All users of Matek F722-STD with any hexacopter mixer.

## Additional context

stronnag, 2018-10-25:
#3955, via RCG discussion

stronnag, 2018-10-25:
addressed by #3955

closed 2018-10-25 by stronnag