iNavFlight/inav #3956: CLI "feature DISPLAY" indicates "Invalid name", Cannot enable OLED.

Sedim, 2018-10-26:
## Current Behavior
When using the "feature DISPLAY" in CLI, CLI returns "Invalid name". 
[image]

It is also enabled in the configuration tab:
[image]

## Steps to Reproduce

1. Select CLI
2. Type "feature DISPLAY"
3. The response is "Invalid name"

## Expected behavior
Expected "Enabled DISPLAY"

## Additional context

https://pastebin.com/cA9W9Kcy

---

INAV/KAKUTEF4V2 2.0.0 Aug 19 2018 / 21:14:39 (dbdd1656a)
- FC Board name and vendor:
Holybro Kakute F4 V2
- INAV version string:
2018-10-25 @ 21:23:03 -- Flight controller info, identifier: INAV, version: 2.0.0

2018-10-25 @ 21:23:03 -- Running firmware released on: Aug 19 2018 21:14:39

2018-10-25 @ 21:23:03 -- Board: KTV2, version: 0

2018-10-25 @ 21:23:03 -- Unique device ID received - 0x3000304648500a20303056

digitalentity, 2018-10-26:
`feature DISPLAY` is not supported on KAKUTE V2 board at the moment

jfd06, 2018-10-27:
Hi,
Same issue for me with a Matek F405 Wing.
I tried 2 differents firmwares:

# version
# INAV/MATEKF405SE 2.0.0 Aug 20 2018 / 19:22:34 (dbdd1656a)
# version
# INAV/MATEKF405SE 1.9.1 Apr 27 2018 / 15:30:33 (03a5c192)

None of the versions allowed me to enable "feature DISPLAY":

# feature list
Available: THR_VBAT_COMP VBAT TX_PROF_SEL BAT_PROF_AUTOSWITCH MOTOR_STOP SOFTSERIAL GPS TELEMETRY CURRENT_METER 3D RX_PARALLEL_PWM RX_MSP RSSI_ADC LED_STRIP DASHBOARD BLACKBOX TRANSPONDER AIRMODE SUPEREXPO VTX RX_SPI PWM_SERVO_DRIVER PWM_OUTPUT_ENABLE OSD FW_LAUNCH TRACE 

# feature
Enabled: VBAT TX_PROF_SEL MOTOR_STOP SOFTSERIAL TELEMETRY CURRENT_METER DASHBOARD AIRMODE OSD 

# feature DISPLAY
Invalid name

However, Matek website still shows the oled display on its wiring diagram together with INav 19.x configuration information and it's also written:
I2C1 [DA1 CL1] supports Baro( BMP280 /MS5611 /BMP085) and OLED(0.96″)

What is the issue ? Does the feature is definitely removed ?

jfd06, 2018-10-30:
Hi,
Ok so to be sure I have just tested my first Oled display (an old Crius...) with an Arduino Nano and Adafruit SSD1306 demo program. Didn't worked, black screen... 
So I tried few minutes ago with the same Arduino setup another Oled display I ordered few times ago from Banggood. This display showed demo drawings as expected. 
So I wired it on my FC put some power in and ... it works !
I remembered I left the configurator with the Oled feature ON. In CLI, feature confirm DASHBOARD is enabled. I removed the feature by typing "feature -DASHBOARD" and saved/reboot. Display was off as expected and in Configuration Tab so the feature is effectively disabled.

So @shellixyz, your right ! Merci ;)

The OLed feature is now called DASHBOARD but the tool tip needs to modified.

Thank you @Dhananjaya96 and @shellixyz for the support !

closed 2018-11-04 by DzikuVx