iNavFlight/inav #3963: Question: Pro and Contra for Compass on fixed wing?

b14ckyy, 2018-10-28:
Hi All,

I just have an important question for the use of a electronic compass on a fixed wing.

I know in the Wiki is mentioned that a compass is not recommended. Anyway I can activate and use it. 
I also activated the compass on my Stealth Wing and it worked quite well. So can somebody explain what the downsides are if it is activated? Or how INAV even uses the compass and if it can have any benefits?

I cannot find any answer to this. 

In my imagination, INAV can use it to get a faster and more precise initial heading to a waypoint when turning. I Have another plane, the Flite Test Bushwacker Foamboard thing, without an compass activated and I experienced often overshoots in turns when heading to the next WP. Never experienced that that much on my wing with compass activated. 

An explanation would be great! Thank you!

shellixyz, 2018-10-29:
On a fixed wing aircraft the mag is used only when the heading given by GPS (course over ground) is not available meaning before taking off when the speed is less than 3m/s. So it is basically useless apart from seeing the direction the plane is pointing to before takeoff.

Overshoot depending on what you mean can be caused by horizon drift, a known issue making the plane turn too much or too small max angle in navigation modes making the plane turn slowly. There is no solution to the first issue for now but for the second one you can increase the max bank angle in navigation modes by increasing `nav_fw_bank_angle`.

On a plane the GPS CoG is enough. It represents the true direction of movement of the plane. The only advantages to use a mag on a plane would be for horizon drift compensation but iNav doesn't use the mag for that at least not yet and also dead reckoning (not done by iNav).

b14ckyy, 2018-10-29:
> So it is basically useless apart from seeing the direction the plane is pointing to before takeoff.

> drift compensation but iNav doesn't use the mag for that at least not yet and also dead reckoning (not done by iNav).

Great thank you. That are the answers I was looking for. So In-Flight it does ignore the compass anyway but it could be an advantage to keep the throw direction during Auto Launch until GPS CoG kicks in. That could explain why the other Plane without Compass sometimes makes corrections during launch from ground and flies a bit sideways, if the first direction Information is not accurate enough and the AR Wing always is rock stable by keeping the throw direction. So as long I have no launch issues with compass I prefer a straight launch and will keep it activated. Thank you :)

closed 2018-10-29 by b14ckyy