iNavFlight/inav #3965: anyfc f7 sd card black box is not working

Dhananjaya96, 2018-10-29:
inav sdcard black box is not working .
## Steps to Reproduce
every time cli command -  set sdcard_detect_inverted = ON

1.first attempt on inav 1.7 and there its not working 

2.then on inav 1.8 ,  1.9.0   , 1.9.1 on this version the fc controller unable to create "freespace" system file keeping in mind that sd card is formatted by sdcard formatter windows application . to solve this problem first I have to flash betaflight firmware .then it  create system file . and then flash inav once again and it works .
3. after first 3 or 4 arming process there it shown on configurator "fatal error reboot to retry" after rebooting it works.
4. same I done with inav 2.0 still is not working it shows " fatal error reboot to retry" and nothing happens 
 5. it might be possible that hardware may be damage but no its not .sdcard works with betaflight and not with inav.

## Suggested solution(s)

inav is best option for autonomous flight but it needs some tuning for that blackbox is necessary .
kindly solve this black box problem 

## Additional context

    https://pastebin.com/C0yX8uat

---

- FC Board name and vendor: ANYFCF7 by banggood
- INAV version string: INAV/ANYFCF7_EXTERNAL_BARO 2.0.0 Jun 26 2018 / 22:30:33 (5db1e1a45)

CcLaFleche, 2018-11-04:
Others F7 too

digitalentity, 2018-11-04:
Fix ready, verified on several F7 boards. Closing.

closed 2018-11-04 by digitalentity

Dhananjaya96, 2018-11-05:
thank you so much