iNavFlight/inav #3967: Enable USE_RX_MSP for Matek F722

shlapion, 2018-10-29:
Greeting, 

I am trying to enable the `USE_RX_MSP` feature for my Matek F722 FC. 
In order to to this I have tried to build my own custom firmware by adding `#define USE_RX_MSP` to the target.h file of the board. 

However when I try to compile I get the following error: `Makefile:341: recipe for target 'obj/main/MATEKF722/cms/cms_menu_battery.o' failed`

Any help is greatly appreciated !

stronnag, 2018-10-29:
It should not be necessary to define `USE_RX_MSP` in the target header file, it is already defined in `common.h`.

You need to provide more of a traceback for us to see why your compilation fails.

shlapion, 2018-10-29:
Thank you for the reply @stronnag .

Indeed it is defined in the `common.h`. However...

Long story short, I am trying to use ROS to send MSP commands to the FC. More specifically I am trying to send RC data using a ROS node. To get the FC to process the data sent I am using the  `MSP_SET_RAW_RC` function. The messages are accepted but not processed. 
After digging around, I found a thread in which someone pointed out that `USE_RX_MSP` has to be enabled for the FC to process the message. So I thought it might be worth a try, and compile my own firmware. 

What exactly could I provide to shed some light on the compilation error?
For starters the terminal output attached below 

make TARGET=MATEKF722
%% target.c 
%% main.c 
%% common_hardware.c 
%% assert.c 
%% build_config.c 
%% debug.c 
%% memory.c 
%% printf.c 
%% time.c 
%% config_eeprom.c 
%% config_streamer.c 
%% feature.c 
%% parameter_group.c 
%% adc.c 
%% bus.c 
%% bus_busdev_i2c.c 
%% bus_busdev_spi.c 
%% bus_i2c_soft.c 
%% display.c 
%% exti.c 
%% gps_i2cnav.c 
%% io.c 
%% io_pca9685.c 
%% light_led.c 
%% logging.c 
%% rx_nrf24l01.c 
%% rx_spi.c 
%% pitotmeter_adc.c 
%% pwm_esc_detect.c 
%% pwm_mapping.c 
%% pwm_output.c 
%% rcc.c 
%% rx_pwm.c 
%% serial.c 
%% sound_beeper.c 
%% stack_check.c 
%% system.c 
%% timer.c 
%% lights_io.c 
%% cli.c 
%% config.c 
%% controlrate_profile.c 
%% fc_core.c 
%% fc_init.c 
%% fc_tasks.c 
%% fc_hardfaults.c 
%% fc_msp.c 
%% fc_msp_box.c 
%% rc_adjustments.c 
%% rc_controls.c 
%% rc_curves.c 
%% rc_modes.c 
%% runtime_config.c 
%% settings.c 
%% stats.c 
%% failsafe.c 
%% hil.c 
%% imu.c 
./src/main/flight/imu.c: In function 'imuMahonyAHRSupdate':
[... 141 lines cut ...]
     OSD_UINT8_CALLBACK_ENTRY("PROF", cmsx_onBatteryProfileIndexChange, (&(const OSD_UINT8_t){ &battDispProfileIndex, 1, MAX_BATTERY_PROFILE_COUNT, 1})),
     ^
./src/main/cms/cms_menu_battery.c:134:5: error: (near initialization for 'menuBatteryEntries[1].type')
./src/main/cms/cms_menu_battery.c:135:5: warning: missing braces around initializer [-Wmissing-braces]
     OSD_SUBMENU_ENTRY("SETTINGS", &cmsx_menuBattSettings),
     ^
./src/main/cms/cms_menu_battery.c:135:5: warning: (near initialization for 'menuBatteryEntries[1].<anonymous>') [-Wmissing-braces]
./src/main/cms/cms_menu_battery.c:135:5: error: initializer element is not constant
./src/main/cms/cms_menu_battery.c:135:5: error: (near initialization for 'menuBatteryEntries[1].<anonymous>.func')
./src/main/cms/cms_menu_battery.c:137:5: error: initializer element is not constant
     OSD_BACK_ENTRY,
     ^
./src/main/cms/cms_menu_battery.c:137:5: error: (near initialization for 'menuBatteryEntries[1].data')
./src/main/cms/cms_menu_battery.c:139:1: error: initializer element is not constant
 };
 ^
./src/main/cms/cms_menu_battery.c:139:1: error: (near initialization for 'menuBatteryEntries[1].flags')
./src/main/cms/cms_menu_battery.c:139:1: warning: missing initializer for field 'text' of 'OSD_Entry' [-Wmissing-field-initializers]
In file included from ./src/main/cms/cms.h:7:0,
                 from ./src/main/cms/cms_menu_battery.c:29:
./src/main/cms/cms_types.h:62:24: note: 'text' declared here
     const char * const text;
                        ^
Makefile:341: recipe for target 'obj/main/MATEKF722/cms/cms_menu_battery.o' failed
make: *** [obj/main/MATEKF722/cms/cms_menu_battery.o] Error 1

shlapion, 2018-10-29:
I am using version 5.4.0. I updated to version 6.5.0 and made it the default compiler. 
Ran make again, facing the same issue.

shlapion, 2018-10-29:
On which version of iNav did you test RX_MSP?

stronnag, 2018-10-29:
6.5 is probably too old, please use 7.3 or 8.2

I've tested RX_MSP on probably every version of iNav. There was one in the distant past where it was broken. You could search the closed issues if it's of historic importance.

stronnag, 2018-10-29:
```
feature RX_MSP
set receiver_type = MSP
save
```
Tested by:

* Configurator showing receiver tab on /dev/ttyACM0 (USB port)
* Running the file http://seyrsnys.myzen.co.uk/msp_set_rc.rb as
  `./msp_set_rc.rb /dev/ttyuSB0`
  i.e. a separate UART enabled for MSP, in this case via a 3DR radio, but a USB/TTY connection would work.

You should then see the configurator bars bounce around and something like:
```
Tx: 1555,1501,1501,1505,1017,1442,1663,1969
Rx: 1501,1505,1501,1555,1017,1442,1663,1969

Tx: 1558,1551,1563,1527,1017,1442,1663,1969
Rx: 1551,1527,1563,1558,1017,1442,1663,1969
```

The reason why the RX values differ in order from the TX values is that this model is set to TAER and I forgot to change it.

Note also that the whole thing is monstrously inefficient (just a POC) and you may need to fiddle around with the `sleep`s .

shlapion, 2018-10-29:
Many thanks !
Will look test it out tommorow first thing in the morning, sitting in a lecture at the moment.

stronnag, 2018-10-29:
Have fun. In my experience while RX_MSP seems to work for a while, it has a number of undesirable effects, inter alia:
* Spams the ground side with the unsolicited (violation of the MSP v1) spec CMS message id 182
* Corrupts the FC, leading to
* Spontaneous reboot.

So really it's not much use. Here's a rather more robust test tool (golang) http://seyrsnys.myzen.co.uk/msp_set_rx.tar.xz. Even if this tool is buggy, it should not crash the FC using a published API.

nik012003, 2018-10-30:
@shlapion  It would be awesome it you could test my MSP_RX python code on your setup.
I confirmed it to work on iNav 2.0 on omnibus f4 pro v3 and SP racing F3.
The code is here : https://github.com/nik012003/OpenGS
Inside Protocol_handler there is a jupyter notebook with some tests, including set_raw_rc via MSP v2.
Thank you in advanced.

shlapion, 2018-11-12:
@nik012003 thank you for the insight and sorry for the late reply. Was away for a couple of weeks. 
Will keep you posted with the results.

shlapion, 2018-11-13:
@nik012003 being new to Python I am not really sure how to run your example inside the Protocol_handler, so I wrote my own script. 

I have managed to read the sensor data(`attitude`,`raw_gps` ,`raw_imu`) works wonderfully.
Setting the RC values using the `set_raw_rc` function works however the copter does not arm. (The values for the channels have been stored while the copter was armed and then passed to the  `set_raw_rc` as arguments)

Using the `get_rc` function I have the same values as set with the `set_raw_rc` function so I'm guessing it works. 

I have enabled the `feature RX_MSP` and set the `receiver_type = MSP` as @stronnag suggested. 

Am I missing something?

Thank you for the support!

nik012003, 2018-11-13:
@shlapion I think you should enable motor control in the Motor part of the configurator to do so.
It is a safety feature which disables arming when the configurator is running.
If you have any problems feel free to contact me on hangouts or skype so we can better understend the problem and fix it faster.

shlapion, 2018-11-13:
@nik012003 thank you for the quick reply. 
I have enabled the motor control in iNav. Sadly still no result. 
Would be great if we could take this conversation forward on hangouts or skype. Whichever works for you better. 
You can drop me a mail/message on google at the same username as here.

nik012003, 2018-11-13:
@shlapion I've added you on skype.

stronnag, 2018-11-13:
> I have managed to read the sensor data(`attitude`,`raw_gps` ,`raw_imu`) works wonderfully.
> Setting the RC values using the `set_raw_rc` function works however the copter does not arm. 

Arming and disarming via RX_MSP work as expected. Just make sure there is nothing blocking arming and you have the right channel map.

shlapion, 2018-11-14:
Have gotten it to work. The rate at which the MSP messages were sent was the issue. 
@stronnag  thank you for the support. 
For anyone else trying to make this work, will clean up the code and share it.

closed 2018-11-14 by shlapion