iNavFlight/inav #3969: ASGARD32F4 MAG

cpusmith, 2018-10-30:
For the ASGARD32F4 if the #define USE_MAG and the rest of the MAG entries are commented out in the target.h file does that mean that magnetometers can't work with the board?

teckel12, 2018-11-02:
@cpusmith The FC doesn't have i2c pins, so it's a physical limitation of the board.

KnuckleUpFPV, 2018-11-12:
You should still be able to use uart 3. I havent found a board without i2c broken out that doesnt work on uart3.

KnuckleUpFPV, 2018-11-12:
Yeah just looked it up. I2c is available on uart3 for your board.

KnuckleUpFPV, 2018-11-12:
https://store.myairbot.com//media/iverve/uploadpdf/1521040968_80095-Asgard32-F4.pdf

shellixyz, 2018-11-12:
UART3 is not shared with I²C on this board. If you can find where the SDA/SCL pads are connected to the MCU on this board we can enable I²C support.

KnuckleUpFPV, 2018-11-12:
I just looked it up. It shows iic which is i2c on uart3

KnuckleUpFPV, 2018-11-12:
Check the link provided. It shows iic on uart3

digitalentity, 2018-11-12:
On ASGARD32F4 UART3 is on pins PC10/PC11, there is no I2C bus available on those pins on STM32F405 processor:

[image]

There is no point in flogging a dead horse here, there is no I2C available on this board.

closed 2018-11-12 by digitalentity

KnuckleUpFPV, 2018-11-12:
I cant post the image from my phone but it clearly says it on the provided link of the manual.

KnuckleUpFPV, 2018-11-12:
SBUS UART1 Hardware inverter
DSM2 UART1 Jumper over the inverter
Smart Audio VTX UART2
Smartport ANY Inverters needed
ESC Telemetry UART4 Connected
IIC UART3
SPARE UART UART5 UART6

KnuckleUpFPV, 2018-11-12:
According to the manufacturer's website the stm32f405 rgt6 which is the chip on the asgard has i2c connected to uart3. This is the datasheet specific to the rgt6. All 180 pages or so for this specific chip.

KnuckleUpFPV, 2018-11-12:
http://pdf1.alldatasheet.com/datasheet-pdf/view/510007/STMICROELECTRONICS/STM32F405RGT6.html

please see page 59

digitalentity, 2018-11-12:
Once again. 

STM32F405RGT6 datasheet (https://www.st.com/resource/en/datasheet/dm00037051.pdf).

page 52:
[image]
UART3 shared with I2C is on pins PB10 & PB11

page 53:
[image]
UART3 is on pins PD8 & PD9. Not shared with I2C

Page 56:
[image]
UART3 is on pins PC10 & PC11. Again, not shared with I2C. This is what we have on ASGARDF4 board.

KnuckleUpFPV, 2018-11-12:
From the link you just posted page 63 says otherwise. Its the same chip used on the matek and dys boards. Dys doesnt have i2c broken out either. Yet its still connected to uart3

digitalentity, 2018-11-12:
@KnuckleUpFPV on ASGARDF4 UART3 is wired to pins PC10 & PC11. Your reference to page 63 is irrelevant. Please stop bragging, you're not going to get I2C working on that board - there is NO hardware possibility unless you solder directly to CPU pins.

KnuckleUpFPV, 2018-11-12:
Its not bragging bud. Its the truth. Its the same chip. It works.

digitalentity, 2018-11-12:
@KnuckleUpFPV if you genuinely belive that it works - feel free to create your own firmware for ASGARDF4.

KnuckleUpFPV, 2018-11-12:
I dont need it. Its the guy that opened this. Just making him aware. If he wants to compile his own version then he is set.

datamann, 2018-11-12:
If you are using target ASGARD32F4 then I2C are mapped to:
// I2C
#define USE_I2C
#define USE_I2C_DEVICE_2
#define I2C2_SCL                PB10
#define I2C2_SDA                PB11

And UART3:
#define USE_UART3
#define UART3_RX_PIN            PC11
#define UART3_TX_PIN            PC10
ref.
https://github.com/iNavFlight/inav/blob/master/src/main/target/ASGARD32F4/target.h

But if there are no pads to connect to then it won't work... unless it's possible to use SWC and SWD pads...

KnuckleUpFPV, 2018-11-12:
Im not misleading anyone. Its the same chip as the matek and dys. The dys isnt broken out, but i2c works on uart 3. Its no different than using a board with them broken out. If you have something on the  i2c pads uart3 becomes unusable. I get it you have to stick up for each other. But it works. Its the same chip. I can show you the same chip on the dys f4 with gps on uart3. The datasheet posted says i2c is available on uart3. I was told the exact same thing about the dys f4. There are no scl or sda pads broken out so no i2c. Then i found an rcgroup post that stated i2c is almost always on uart3. Low and behold its connected and works. Just like you can use the scl and sda pads on the matek as uart3 and vice versa. This chip supports i2c on uart3.

KnuckleUpFPV, 2018-11-12:
The uart is connected to pb10/11, pc10/11. You can use i2c because the uart is on all 4 pin assignments.

digitalentity, 2018-11-12:
You don't seem to get the simple fact that there is more than one option to
connect UART3 to the CPU, but not all of those options allow UART3/I2C
sharing.

On OMNIBUS and variants it's possible to share, on ASGARDF4 it's not
possible to share. All complaints go to CPU manufacturer.