iNavFlight/inav #3970: GPS data delayed several seconds

FranzAr, 2018-10-30:
iNav 2.0
Airbot Omnibus F4 pro V3
Beitian BN-220 GPS

GPS data is delayed up to 10 seconds as you can see in my video. This doesn't happen in every flight but every one and then. If it happens it is during the whole flight.

https://www.youtube.com/watch?v=o8yy23SOid0

I'm not sure if this is an iNav or a hardware issue. Over in the facebook group 1 or 2 other users had similar issues. 

Any hints?

digitalentity, 2018-10-30:
Duplicate of #3908

closed 2018-10-30 by digitalentity

konraddi, 2018-11-06:
This issue existed since at least August 2017! I was never able to use those two gps modules because of that. Gladly I kept them for more than a year, an tested them this morning with 2.0.1, working properly now without delay, thanks for fixing!! #1896

https://github.com/iNavFlight/inav/issues/1896

konraddi, 2018-11-06:
[
[image]