iNavFlight/inav #3972: Home Heading Deviation / Home Distance in Telemetry

FPVZaphod, 2018-10-31:
## Current Behavior
Currently the difference between Home Course and actual Course is not sent by telemetry

## Desired Behavior
Some Setups (like Connex Prosight) don't support OSD, therefore it's not possible to display a Home Arrow and DTH in the Video Stream. It would be helpful to have this data in your Radio available. It could tell you by voice output the actual home heading deviation.  As an example, 90 can tell you that Home is 90° to the right, or -90 that Home is 90 to the left. Same with home distance.
Would be nice if also available with Crossfire Telemetry.

## Suggested Solution
Two new telemtry values:
Heading Home Deviation ( -180 to 180 )
DTH / Distance to Home ( Meters )

## Who does this impact? Who is this for?
- Setups without OSD, like digital ones.
- Groundstation based OSDs to insert OSD between Video RX and Goggles/Screen

teckel12, 2018-11-02:
@FPVZaphod What transmitter are you using? If Taranis, you can just use a script to get this information. Probably with other transmitters too.

FPVZaphod, 2018-11-05:
@teckel12 : Taranis, but with Crossfire. Never managed to make them work, while with a FrSky-RX it's not a problem. 
Just thought it's a small thing to send this two values down...

teckel12, 2018-11-05:
@FPVZaphod The INAV Lua Telemetry script is adding support for Crossfire (the current limitation is that Crossfire is not exporting data Lua Telemetry needs to function). The next release of INAV will add this information at which time you'll be able to use Lua Telemetry to get the information you're looking for.

FPVZaphod, 2018-11-20:
Cool, thanks!

closed 2019-02-02 by shellixyz