iNavFlight/inav #3973: No pull-down on PWM5 on omnibus F4 pro

BlueSkyLark, 2018-10-31:
## Current Behavior
Omnibus F4 Pro V3 
Inav/omnibusf4pro V2.0.0
Outputs pulse from PWM5 but only between 2v and 3.3v (approx). PWM 0-4 and PWM6 work ok (0-3.3v).

## Steps to Reproduce
1. Quad setup. Added mixer from rc channel to drive gimbal pitch input to BCG board.

## Expected behavior
BCG does not respond to output, but servo on same pins works okay. On investigation I find that the signal is not pulled down to ground (servo presumably has internal pull-down where BCG does not). I don't have this problem on any of the other PWM outputs.

## Suggested solution(s)
My guess is that the pull-down (BRR?) bit is not set - probably related to the use of PWM5 for LED strip on the OMNIBUSF4PRO_LEDSTRIPM5 build (which I'm not using).

## Additional context
Okay, this is a minor niggle, and easily fixed by the addition of a pull-down resistor on the BCG board, but I wanted to post it here so in case anybody else has the same problem. My actual work-round was to add an extra mix so that I can connect the gimbal to PWM6, which works as expected. Of course, it's always possible that I have a faulty FC where the pull-down is just burned out, but I don't think so.

---

Omnibus F4 pro V3 (Chinese clone from gearbest).
INAV/OMNIBUSF4PRO 2.0.0 Aug 19 2018 / 21:27:27 (dbdd1656a)

digitalentity, 2018-10-31:
Motor outputs are configured push-pull. You have faulty hardware.

BlueSkyLark, 2018-10-31:
Thanks digitalentity. I'm sure that you're right. It will be my hardware. Sorry for the time wasting.

closed 2018-10-31 by BlueSkyLark

digitalentity, 2018-10-31:
No worries @BlueSkyLark, it never hurts to ask 😄