iNavFlight/inav #3976: RTH Failsafe - If no gps lock simply drop.

simtcrom, 2018-11-01:
## Current Behavior
Currently, If failsafe is set to RTH in inav  and if no gps lock found and turn off TX, motors continue spinning.

## Desired Behavior
If failsafe is set to RTH in inav  and if no gps lock found simply stop the motors.

## Suggested Solution
I dont have a solution, its just feature request,

## Who does this impact? Who is this for?
This impact failsafe. It is for any users.

## Additional context
None.

teckel12, 2018-11-01:
It could be debated that "land" should be the fallback. Maybe a fallback option instead?

simtcrom, 2018-11-02:
> It could be debated that "land" should be the fallback. Maybe a fallback option instead?
Sure. That works.

simtcrom, 2018-11-02:
> It could be debated that "land" should be the fallback. Maybe a fallback option instead?

Sure. That works

kardon18, 2018-11-05:
_@teckel12 
It could be debated that "land" should be the fallback. Maybe a fallback option instead?_

I have thought about variations of Failsafe fallback for Quads, more than for Fixedwing.
But this one does depend on having a GPS lock.

_My idea would be to have the FC place the quad into current **Altitude-PosHold** when **Failsafe** is triggered, for a configurable period of time.  To give the pilot some time to move the TX antenna around and possibly re-establish the RC signal lock again.
Once the Altitude-PosHold time has expired, and if the **Failsafe** condition still remains. Then the quad would proceed to RTH._

My reason being. Is its good to have a Failsafe/RTH fall back on a quad. But it is difficult to know how to implement a specific "**Altitude Control Mode**". Due to the places that a 2.4G Failsafe could occur. 
e.g. under a tree. This would be a problem, because it would proceeded to fly up into the branches when trying to returns to home. 

To have the quad to Land or Drop, defeats one of the primary reasons for having a GPS on board.

hali9, 2018-11-05:
> Currently, If failsafe is set to RTH in inav and if no gps lock found and turn off TX, motors continue spinning.

I think emergency landing was called, which uses Land settings, so try set failsafe_throttle = 1000.
This should disable motor in emergency landing

closed 2019-01-18 by stale[bot]