iNavFlight/inav #3983: iNav 2.0 VTX settings don't save

craig34a, 2018-11-04:
Boxcam VTX has no half duplex / tramp features. 
OSD shows "Invalid settings vtx_power"
Craft is not arming
Changing settings via CLI  does not save. 

Dump shows: 

set vtx_halfduplex = ON
set vtx_band = 3
set vtx_channel = 1
set vtx_power = 5
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5705
set vtx_pit_mode_freq = 0

Trying to change these vtx settings will not save. 
set vtx_low_power_disarm =OFF 
for example does not result in any change. 

These settings prevent arming of craft. 

What to do? 

Flight controller F-35 lightning
VTX Boscam TX 832 5.8GHz 600 mW

teckel12, 2018-11-05:
I believe the solution is to `set vtx_power = 4` or compile the development build, as it's been fixed in the next release.

gregm412, 2018-11-05:
@teckel12. I have the same problem with the f35 and the furious stealth vtx. Can you compile the development build for the f35 or point me in the right direction to do so. Thank you.

teckel12, 2018-11-05:
@gregm412 Setting `vtx_power` to `4` would at least work and verify the development HEX would fix the problem.

teckel12, 2018-11-05:
@gregm412 Also, it's always nice to give the EXACT target name when you want someone to compile something for you.  There's so many FC and variations that many times I build a HEX only to find out it's not what you want.  Always specify the exact target.  In this case, I can only assume you mean the `FF_F35_LIGHTNING` target?

[inav_2.1.0_FF_F35_LIGHTNING.hex.zip](https://github.com/iNavFlight/inav/files/2550752/inav_2.1.0_FF_F35_LIGHTNING.hex.zip)

DzikuVx, 2018-11-11:
fixed in #4008

PR #4008 (2018-11-12): allow vtx_power 5 to be considered as a valid setting
Apparently vtx power 5 is valid for Tramp protocol but was considered as invalid value by INAV itself. This allows power level 5 to be used again

this fixes: #3983 #3938 #3924
closed 2018-11-11 by DzikuVx