iNavFlight/inav #3986: INAV PPM receiver AETR channels only

tomasssk, 2018-11-04:
I have have a problem using SP F3 combined with FS IA6. In the receiver tab, only the AETR channels are changing their state when I am using the transmitter, but the rest of the aux channels 5-8 is not reacting.
The receiver is connected to the FC through IO1 and IO2, when I check with multimeter I see that the input to the aux channel pins in the IO1-2 is changing accordingly.
Before I flashed INAV firmware (I have the latest 2.0.0) I was using both cleanflight and betaflight and there it was working fine.
Any ideas what I might be doing wrong?

hali9, 2018-11-05:
This is not the answer to Your question, but maybe You should try this:
http://endoflifecycle.blogspot.com/2016/10/flysky-ia6-ibus-setup.html

hali9, 2018-11-05:
SPRACINGF3 target.c
```
{ TIM3,  IO_TAG(PB4),  TIM_Channel_1, 0, IOCFG_AF_PP, GPIO_AF_2,    TIM_USE_PWM | TIM_USE_MC_SERVO },   // RC_CH5
{ TIM3,  IO_TAG(PB5),  TIM_Channel_2, 0, IOCFG_AF_PP, GPIO_AF_2,    TIM_USE_PWM | TIM_USE_MC_SERVO },   // RC_CH6
{ TIM3,  IO_TAG(PB0),  TIM_Channel_3, 0, IOCFG_AF_PP, GPIO_AF_2,    TIM_USE_PWM | TIM_USE_MC_SERVO },   // RC_CH7
{ TIM3, IO_TAG(PB1), TIM_Channel_4, 0, IOCFG_AF_PP, GPIO_AF_2, TIM_USE_PWM | TIM_USE_MC_SERVO }, // RC_CH8
```
And in pwm_mapping.c:
**// Handle outputs - may override the PWM/PPM inputs**
```
            // Multicopter
            if (init->useServoOutputs && (timerHardwarePtr->usageFlags & TIM_USE_MC_SERVO)) {
                type = MAP_TO_SERVO_OUTPUT;
} 
```

tomasssk, 2018-11-05:
> So if Ypur mixer use servos (airplane, tricopter, quad with gimbal, ...) RC 5-8 will overwrite by servos.

No, I don't use servos at all. But when i go to the "servos" tab, there is actually one already inserted (servo0) and I can't remove it...see the screenshot.
[image]
Should this one be removed? I am very sure it was there by default, as I have never been using "servos" tab...

tomasssk, 2018-11-05:
Ok, solved...unintentionally I have added a servo in the "mixer" tab. It works fine after removing it.
Thanks a lot for help!

shellixyz, 2018-11-08:
Fixed, closing.

closed 2018-11-08 by shellixyz