iNavFlight/inav #3987: FF F35 INAV F4 Board with MPU9250 MAG ISSUE

maRCnh33, 2018-11-05:
I have an INAV f4 (furious fpv FF35 clone) and the mpu9250 chip contains the gyro, acc and mag. The mag is placed.... this problem is already discussed here on github but i still don't know what to do! When i do a 90 degree flip the heading goes in the opposite direction. Quad is moving (yaw axis) to the left, the mag reading goes to the right! it's actually this story: https://github.com/iNavFlight/inav/issues/590#issuecomment-250940248
i hope someone can help me, i am struggling for days now..... 

Please help!

maRCnh33, 2018-11-05:
it would be nice if there is an on/off option for ivert x, y or/and z axis!! then its easy..... enable x-axis mag reading.... and then the yaw goes left... and the mag reading to left..... i hope there will be a day that this issue will be solved.... yesterday i found the 2.0.1 firmware, install it.... but.... the same issue!

digitalentity, 2018-11-05:
Onboard compass is a bad idea, especially when FC is coupled with a PDB

maRCnh33, 2018-11-05:
yes i know..... but PH is perfect only pointed in one direction..... so, if there is magnetic interfence that isnt possible...

maRCnh33, 2018-11-05:
this is the problem : https://github.com/iNavFlight/inav/issues/590#issuecomment-435973016

digitalentity, 2018-11-05:
If PH is working only when compass is pointed in one direction - issue is 100% interference. Axis alignment issue is fixed if I remember correctly.

maRCnh33, 2018-11-05:
if there is interference then there is ALWAYS a toiletbowl effect... in all directions....

the alignment of the mag sensor can be changed now in inav, thats correct. But when the board is flipped left will be right and right will be left.... so when i rotate the board (yaw) to the left...the quad is rotating to the right....

maRCnh33, 2018-11-05:
this is exactly the problem:

As the issue looked like a misaligned sensor I finally found what needs to be done mag y- or x- axis needs to be mirrored.
I have not yet been flying - but watching the model on the status screen - I expect that fixes the compass issue. Is the MPU9250 - or the AK8963 potentially not correctly factory trimmed or simply inverting 1 axis?

I did that in the compass_ak8963.c file:

magData[X] = -(int16_t)(buf[1] << 8 | buf[0]) * magGain[X];
//line reinserted with *1 / *-1 below: magData[Y] = -(int16_t)(buf[3] << 8 | buf[2]) * magGain[Y];
magData[Z] = -(int16_t)(buf[5] << 8 | buf[4]) * magGain[Z];

#if defined(USE_SPI) && defined(MPU9250_SPI_INSTANCE)
//inserted line not inverted
magData[Y] = (int16_t)(buf[3] << 8 | buf[2]) * magGain[Y];
// cache mag data for reuse
memcpy(cachedMagData, magData, sizeof(cachedMagData));
state = CHECK_STATUS;
lastReadResult = true;
#else
//inserted line *-1 as original
magData[Y] = -(int16_t)(buf[3] << 8 | buf[2]) * magGain[Y];
lastReadResult = ak8963SensorWrite(AK8963_MAG_I2C_ADDRESS, AK8963_MAG_REG_CNTL, CNTL_MODE_ONCE); // start reading again
#endif

maRCnh33, 2018-11-05:
https://github.com/iNavFlight/inav/pull/3447#issue-197213080

digitalentity, 2018-11-05:
If there is interference - there is always at least one direction where compass shows correct heading. In that direction PH will work perfectly, no toiled bown effect. In all other directions - it won't work.

If you believe that your patch works - verify it and submit a PR.

digitalentity, 2018-11-05:
So far this is clearly a support issue - usage of onboard compass in presence of a strong magnetic interference.

maRCnh33, 2018-11-05:
its the different in alignment of the acc/gyro and the mag...

http://www.invensense.com/wp-content/uploads/2015/02/PS-MPU-9250A-01-v1.1.pdf
when you see page 38 of this pdf you can see the different in alignement.. and thats the big issue

maRCnh33, 2018-11-05:
in the future i only buy gps WITH mag!! 😁

datamann, 2018-11-05:
@maRCnh33 
Could you please send me your compass_ak8963.c file?
I have the same board and would like to try your changes...

maRCnh33, 2018-11-05:
[compass.zip](https://github.com/iNavFlight/inav/files/2550456/compass.zip)

tnx!

maRCnh33, 2018-11-06:
Today i downgrade from inav 2.0.1 to 1.9.1..... and.... no problem, i didnt make any change to alignment of mag.... on the inav GUI screen its working normally.. when i turn the fc to the right the heading change also to the right!!! when i do the same in inav 2.0.0 or 2.0.1, turn the fc right the heading goes left!... so... i think its a bug in the new INAV.... on the bench it works fine now, tommorow i will do a testflight and test PH in all directions... we will see (i hope this is the final test and everything will work....

datamann, 2018-11-06:
Nice find :-) I haven't had the time to test this yet but I will...

maRCnh33, 2018-11-06:
Tommorow a testflight with ph.... crossed fingers!

datamann, 2018-11-06:
I looked at 1.9.1 code and the file compass ak8963c has not been changed since.

maRCnh33, 2018-11-06:
i bought this fc at Banggood.... i just look at the productpage... and there on the bottom of the page is a question&answer page.... this is the text (its finally clear now, only use 1.9.1 for this board:  >>>>>>>>>>>>>>>>>
Q :That iNAV firmware must be loaded?
A :
INAV/FF_F35_LIGHTNING 1.9.1 Apr 21 2018 / 13:31:58 (03a5c1922), special version done especially for the FC. Not possible to upgrade with newer version.
Answered by benoit35690 on June 13,2018

https://www.banggood.com/INAV-FLIGHT-F4-Flight-Controller-Built-in-OSD-Battery-Voltage-Current-Monitor-For-FPV-RC-Airplane-p-1283459.html?rmmds=search&cur_warehouse=CN

testflight tommorow.... i expect its working...

tnx alot for helping me!!

maRCnh33, 2018-11-08:
yesterday i have test the quad... and.... not good! Some later rain start and i had to stop. Today i have test all 8 options for mag alignment but i didnt work.... when its on 'default' quad PH works only in SOUTH direction.. when i change the alignment to 270flip it has only good PH in NORTH direction.... my battery is always on the top... but i want to test it with battery on bottom of quad.... and now..... PH didnt work in ANY direction (also 270flip setting)...... my final conclusion is that magnetic interference is the problem. It is a 285size vortex .... i have only 1 12v wire from the back to the front for the 2 front esc's, i have twist that wire very tighten.... but its just not enough i think. I have some MU-Metal for shielding magnetic fields. I will put a piece of that stuff on the side of the FC where the 12v wire is... i dont know if it helps... but i have to try, otherwise i dont know! and yeah, a external mag is much better but i dont know for sure if it is possible on my INAV FLIGHT F4 Flight Controller Built-in OSD & Battery Voltage Current Monitor For FPV RC Airplane..... what i think it's i2c connection is on uart6, uart 6 'is also uart3, my rx is connected to uart3.... so, i can by a mag sensor board but i still dont know if i can connect it to this board...... pffff.......little tired of all testing!! but, i will go on.......

maRCnh33, 2018-11-08:
[image]
[image]
[image]
[image]

maRCnh33, 2018-11-10:
[image]
[image]

i have made a shield with 05mm thick MU-Metal, above the fc to prevent interference from lipo battery... and on the side where the 12v power cables are..... and as an extra an extra mumetal sheet over the wire (so double protection) i have also replaced the esc's further from te fc. I also change the side of the power wire, now on the other side further from the mag sensor..... i hope this will work, i will wait now for better weather and i will do a test! i will plce a little more pictures so you can see the difference....

digitalentity, 2018-11-11:
MuMetal will shield all magnetic fields, including Earth's natural magnetic field. Shielding your FC with MuMetal will not fix your issue. 

Dangers of using built-in compass was explained many times in many places. I'm going to close this issue to avoid distracting the developers. Please use [other](https://t.me/INAVFlight) [support](https://www.facebook.com/groups/INAVOfficial) [channels](https://www.rcgroups.com/forums/showthread.php?2495732-Cleanflight-iNav-(navigation-rewrite)-project)

closed 2018-11-11 by digitalentity

maRCnh33, 2018-11-18:
I have removed the INAV F4 FC and install an old naze f1 (also with onboard mag).... no shield, same wiring.... and the gps hold works perfect!!!..... so, 
I think the f4 board is interfering with its own compas...... tnx for all help guys! btw.. i have installed inav 1.7.3 on the naze32