iNavFlight/inav #3989: Give command to CLI: set i2c_speed = 200kHz save -- CPU Load 600%

Dimsimin, 2018-11-05:
## Current Behavior

## Steps to Reproduce

1. SP Pro Racing F3 (Deluxe), baro and magnitometer is not working 
2. give command to CLI:
set i2c_speed = 200kHz
save
3. baro and magnitometer is working 
4. CPU Load 600%

## Expected behavior

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor: INAV/SPRACINGF3
- INAV version string: 2.0.1

digitalentity, 2018-11-05:
This is expected, on SPRACINGF3 all devices are connected to the same I2C bus. Lowering I2C speed slows down communication to all devices (GYRO, ACC, BARO, MAG). You need to use slower looptime.

closed 2018-11-07 by digitalentity

vladimirerdman, 2020-06-09:
For my first experience with drones I bought SPRACINGF3, and I've just upgraded the firmware to "iNav SPRACINGF3 2.5.0" and after rebooting of "iNav 2.5.0 RC2 Configurator" I found that "CPU Load" was about 800%. I went to CLI mode and entered the command:

get i2c_speed

This command shows that i2c_speed was set to 400kHz and it gave me options that I can choose. So, to change speed we use next command:

set i2c_speed = 800kHz

and to save it:

save

After rebooting "CPU Load" became 35% and now everything is great.

Just for information:
To erase the firmware from SP RACING F3 DELUX I use "flash_loader_demo_v2.8.0".
If your computer don't see your SP RACING F3 DELUX by USB then just install driver "CP210x (Windows Drivers)" and use "flash_loader_demo_v2.8.0" to erase the firmware and upgrade it. Don't forget to use bootloader mode.

I hope that this information will be helpful to someone.