iNavFlight/inav #3991: servo for camera/servo pour caméra

diiity, 2018-11-06:
Hello,

I have a MATEK F405 SDT card with INAV 2.0.

I would like to rotate the camera in the up and down direction with a potentiometer.

I soldered on the S6 pin but nothing happens, I think it is necessary to activate a uart or sofserial but in what configuration?

cordially

Diiity
[image]

Bonjour,

Je possède une carte MATEK F405 SDT  avec INAV 2.0.

Je voudrais faire pivoter la caméra dans le sens haut et bas avec un potentiomètre.

J'ai soudé sur le pin S6 mais rien ne se passe, je pense qu'il faut activer un uart ou un sofserial mais dans quelle configuration?

Cordialement

Diiity

wx4cb, 2018-11-06:
no you need to setup in the mixer. Here's how i have my pan servo setup in my mini talon. It's on channel 7, the other channel is the one for my flashing lights. don't remember exactly how it shows up in the mixer tab, but if you poaste these in, the cli and save then go back to the mixer tab, you'll see the two at the bottom say something like "RC Channel 6" "RC Channel 7"

Then on the taranis i just setup another input etc for the slider

# servo mix
smix 0 2 0 -100 0
smix 1 3 0 -100 0
smix 2 4 1 100 0
smix 3 5 1 -100 0
smix 4 4 2 100 0
smix 5 5 2 100 0
smix 6 6 10 100 0
smix 7 7 11 100 0
 
# servo
servo 2 1000 2000 1476 -100
servo 3 1000 2000 1476 -100
servo 4 1000 2000 1476 100
servo 5 1000 2000 1522 100
servo 6 1000 2000 1422 -100

wx4cb, 2018-11-06:
Also, i believe that if you have soft serial set, then S6 is used for that so you might want to find a different servo channel

wx4cb, 2018-11-06:
you are using the quad mixer which im not sure about, i'd have to defer that to someone else, what you could try is pasting in what my servo mix is and see what happens. it's not going to hurt as you can always just reload the mixer

diiity, 2018-11-06:
as the servo is noted s1 in the servo tab, I connected it to S1 of the F405 but it still does not move. The mixer motor 1, 2 etc ... used for what?

wx4cb, 2018-11-06:
what aircraft are you running? a quad ? a fixed wing?

diiity, 2018-11-06:
Sorry, I forgot to specify it, it's a quadri.

I just tried by removing all the mixer motor, in this case there is more engine.

diiity, 2018-11-06:
Can I send you the doc of the F405 SDT?

stronnag, 2018-11-06:
For multirotors, the firmware defines a single servo, on S1 with `inav_x.y.z_MATEKF405.hex` and S6 with `inav_x.y.z_MATEKF405_SERVOS6.hex`.

diiity, 2018-11-06:
Ah, I do not know what firmware is in my F405 SDT.

But if I put on S1 or S6, I have nothing

diiity, 2018-11-06:
Here in doing "version"
INAV/MATEKF405OSD 2.0.0 Aug 20 2018 / 19:20:20 (dbdd1656a)

stronnag, 2018-11-06:
Wrong firmware then ....

diiity, 2018-11-06:
What should be the good version?
And is it possible to change it?

stronnag, 2018-11-06:
That you were told a few posts above https://github.com/iNavFlight/inav/issues/3991#issuecomment-436350980

Just use the servo6 version (with the servo on S6). You need to reflash the board.

diiity, 2018-11-06:
Ok thanks for your help.

I change the firmware and I want to know that I can

diiity, 2018-11-06:
> On vous a dit quelques articles au-dessus de [# 3991 (commentaire)](https://github.com/iNavFlight/inav/issues/3991#issuecomment-436350980)
> 
> Utilisez simplement la version servo6 (avec le servo sur S6). Vous devez reflasher le tableau.
> 
> Si vous avez une carte STD, n'utilisez PAS le micrologiciel OSD obsolète.

I'm sorry but I do google translation that's why I did not understand the first time