iNavFlight/inav #3994: RTH in INAV Version 2.1

Rclay11541, 2018-11-07:
BOARD: Omnibus NXT F7
GPS: BN-880

In INAV 2.0 RTH works well. In INAV 2.1 RTH does not work. No variables changed between flashes.

In 2.1 first action is to increase altitude to random heights. Sometimes this is at full throttle, other times its within 300CM/S parameter.

In 2.1 second action is to turn and fly back at altitude. Flight path is NOT toward home and is instead at random slow straight paths.

Held all variables static and reflashed to INAV 2.0 and RTH works flawlessly again.

digitalentity, 2018-11-07:
Any logs of the problem? Also, there is no version 2.1

Rclay11541, 2018-11-07:
No logs as per issue #3916.

Regarding the version, sorry I think its V2.0.1 released on 11/4/18.

closed 2019-01-20 by stale[bot]