iNavFlight/inav #3999: Kakute F4 V2 use motor 2 as servo

holoduke, 2018-11-08:
I want to use motor 2 as a servo, but this is not working somehow. In the mixer screen you can see id:2 not having a servo assigned.
Motor 3 and 4 are working ok. But if you want to fly a plane with additional elevons you need that extra servo. Is there a way to remap motor 2 to work as a servo?

shellixyz, 2018-11-08:
Short answer: you can't

Long answer: it is possible by making a custom build with M2 reconfigured as a servo output but then you need to use 50Hz for all the outputs

Best solution: Use a PCA9685 board for driving up to 16 more servos through I²C or just use a FC suitable for your application.

Tygrys-1, 2018-11-10:
> Best solution: Use a PCA9685 board for driving up to 16 more servos through I²C or just use a FC suitable for your application.

Can we really have 16 servo channels that way? Some time ago it was limited to 8 AFAIK.

shellixyz, 2018-11-12:
@Tygrys-1 Should be. The chip and drivers supports 16 servos

Szpero, 2018-11-20:
Outputs are related to servo ID's and are affected by mixer?
Or it is just direct pass through RX channels to controller?