iNavFlight/inav #4001: Omnibus F4 pro v3 GPS/Compass/SmartPort and Trump Telemetry.

chief2811, 2018-11-08:
Hi everyone. 

I recently bought an Omnibus F4 pro v3 FC (no clone) in order to hook up a gps and using inav. So I also bought a m8n gps /w compass. 

Cause I’m using a xsr rx I need to have SBUS. Now I’m wondering how to connect also trump telemetry and also smartPort to my fc. 

I was using a F4 non pro fc before that pro. There I was able to map the unused pwm6 pin to softserial for smartport. And i hooked up tramp telemetry to uart6 tx. On pwm5 I use the LED’s.
So everything was fine for now.

 Now I thought I can buy a pro fc because of having a baro included and doing the same thing. Instead of hooking up tramp t to uart6 tx I would use port mapping and use the pwm5 Pin because the pro version have a separate led pin. BUT, my researches show me that port mapping in iNav is not supported right now and I may need a custom fw.

As I understand so far I need to have 2 free UART for hooking up GPS and compass? 

Is this true? Or is there any other possibility to have everything working together. 

So to sum up. I want to hook up:
-SBUS (XSR)
-smartport
-tramp telemetry
-gps
-compass
-LED
-buzzer

For using with inav.

I think in Betaflight I can manage everything to work. Because I can use pwm5 and 6 for softserial and hook up smart port and tramp t to that pins. But then I do not have inav. That actually is my target (pos hold, 3D lock, rth, loiter e.g.) with Betaflight right now just a return to home function is available. 

I hope someone could help me with that issue cause I don’t want to buy another FC so quickly. 

Thx in advance!!!

teckel12, 2018-11-08:
@chief2811 There's a soft serial port on the Omnibus F4 Pro. It's on two small pads. See the Omnibus F4 target documentation for INAV for **VERY** detailed information on this subject (including lots of pics).

Also, for clarification, the compass uses i2c but it uses UART3. So, it doesn't technically use UART3 but it does use the pins that UART3 uses, so it's really about the same as far as port allocation goes.

Finally, yes, INAV doesn't do pin reassignment nor will it anytime soon, so don't consider that as an option at all at this point.

chief2811, 2018-11-09:
I found these to solderpads on ch5 and ch6. But as I figured out there ist ch5 tx und ch6 rx , or the other way around. I don’t remember right now. But for smart port and tramp telemetry I need two tx pins. Right?
Can I use these pins? Right out of the box or do i have to have a custom firmware? 

2nd question. You say that the compass uses i2c but it’s connected to uart3 pins. Is there a possibility to use uart3 and i2c at the same time?

teckel12, 2018-11-09:
@chief2811 Another option would be to flash your Rx and use F.Port instead of SBUS. This would allow you to do both F.Port and S.Port telemetry on the same UART, freeing up the soft serial port.

chief2811, 2018-11-10:
> > Is there a possibility to use uart3 and i2c at the same time?
> 
> No it is not possible.
> 
> The Omnibus F4 doesn't have enough UARTs for what you're trying to do unless you use a Naza GPS which combines GPS+Mag on one serial port. Then you could use S.Bus on UART1, GPS on UART3, S.Port on UART6, Tramp on softserial (CH6)

I do have a Naza gps. But I don’t want to cut the wires. I need a spare CAN cable to build me an adapter. Naza is expensive right here.

chief2811, 2018-11-10:
> @chief2811 Another option would be to flash your Rx and use F.Port instead of SBUS. This would allow you to do both F.Port and S.Port telemetry on the same UART, freeing up the soft serial port.

I heard about this option too. But didn’t go deeper this way. Because it’s written even on frsky Homepage that this is beta. 
But as I understand you with f port you don’t use uart for serial rc anymore? Or is it just possible to have SBUS and smartport at the same uart?

What about ch6 question? I managed to get running everything exapt smartport; on Betaflight. So I used softserial ch6. This small solderpad. I was searching the web for just 3-4 hours to figure out I have to type cli ‘resource SERIAL_TX 11 CH6’ in order to get it working. So I have UART1 SBUS, uart3 magnetometer, uart6 gps, softserial ch6 tramp T.
At the moment I don’t get a gps fix. But I think I wired something not correct. So the question is can I flash inav to the board? And have it the same configuration? Or do i have to have a custom firmware in order to get softserial an ch6?

teckel12, 2018-11-10:
@chief2811 With F.Port control and telemetry are on the same UART. But, they would need to be uninverted as the F4 can't invert and the SBUS inverter on your FC is almost surely a one-way converter. But, I believe the R9 Slim+ exposes uninverted signals. And F.Port is stable, it's been noted as beta for a long time and many people have flown it for a long time without issues.

chief2811, 2018-11-10:
Thanks everyone. For the response.

I think I’ll give inav a second chance right now. I can live without smart port at the moment. Because I’m not into PID tuning so hard. I more like to change VTX settings remotely instead of pushing a tiny little button....

> @chief2811 With F.Port control and telemetry are on the same UART. But, they would need to be uninverted as the F4 can't invert and the SBUS inverter on your FC is almost surely a one-way converter. [...]

That means you have to do another inverter hack with your RX? As long as  you do not have a r9 slim+?

teckel12, 2018-11-11:
@chief2811 Yup, which is why a R9 slim+ or an F7 board are desirable. Also, SmartPort telemetry is much more than adjusting PIDs, I don't even do that but find telemetry required.

chief2811, 2018-11-11:
> @chief2811 Yup, which is why a R9 slim+ or an F7 board are desirable. Also, SmartPort telemetry is much more than adjusting PIDs, I don't even do that but find telemetry required.

I was looking forward to get one of these F7 Boards. I’m interested in Holybros F7 Kakuten. Because I found some sites which explain a lot in detail about this board. The Omnibus F4 pro i will put in a flying wing then.

Droid404, 2018-11-11:
hey guy's noob here not trying to hijack but you 2 seem to know allot with what i need help with. i have a r9mm rx  with  x-lite  r9lite tx  and f4 v3 pro  . what do i need to do to get rssi ? rssi goes over sbus if not analog correct ?  i'v never setup rssi  or  smart port . sbus is setup on j7 and have it flying but no rssi in osd .  i was going to add a gps to j10 tx/rx3 . then were would i hook up smart port? pmw5/6? then remap.. any help or general info would be awesome

teckel12, 2018-11-11:
@Droid404 No resource remapping in INAV.

Typically, you connect SmartPort to the soft serial connection (which is ch5/6, and **NOT** the same as pwm5/6).  This subject is well-traveled.  Please see the following very extensive docs:

https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20Omnibus%20F4.md
https://github.com/iNavFlight/inav/blob/master/docs/Telemetry.md

closed 2019-02-02 by shellixyz