iNavFlight/inav #4007: Inav with xsrf40 can read GPS telemetry on transmitter but not in inav

Rapidlaser, 2018-11-11:
I set up my xsrf40 with frisky smart port GPS it works perfectly in the frisky XD9+ but does not show in inav configurator, so I purchased the M8N pixhawk and set it on uart3 and got exactly the same result, reads fine on the X9D+ but nothing shows in inav configurator, the GPS icon at the to is always red and no messages seen in the GPs page, Can anyone point me in the right direction on how to solve this, I have scoured the net and found nothing.

digitalentity, 2018-11-11:
Please describe how your GPS is connected.

Rapidlaser, 2018-11-11:
Apologies for posting in the wrong place. I have 2 xsrf40 one with the M8N and it is connected to the uart3 port +, - rx, tx, uart3 in inav is set to GPS and GPS is switched on in configuration with UBLOX, ground assistance set to autodetect. the other is the FRsky smart port GPs and it is connected to the smartport, I cannot set it in ports as there is no way to do that. As I explained they both send telemetry to the Frisky Transmitter no problem at all but I cannot get anything in inav, no messages received.

teckel12, 2018-11-11:
If you have a FrSky GPS connected to the Rx, you won't get anything in INAV as you're not connecting it to the FC at all.

For the M8N, if your getting GPS on your Taranis, it must be working on your FC unless you have it connected in a very strange way.

I think you need to break this into two totally different issues. For thr FrSky GPS you're just expecting something that simply doesn't work. For the M8N, that could be diagnosed further with more detailed information as what you're saying doesn't make sense.

Rapidlaser, 2018-11-11:
Hi teckel12, thank you for your reply, I don't understand what you mean by it doesn't make sense, I have soldered the four wires from the M8N to the four pins on UART3 on the xsrf40 as per the instructions +5v, ground and the RX and TX pins, I trust you are aware the xsrf40 is a single board with receiver and Fc on one board? I have followed the instructions that many people have posted on youtube and as explained it works fine on the TX, please tell me what doesn't make sense and I will see if I can get it to make sense. I guess my lack on understanding how these boards work is hampering my progress, I have attached an image of the xsrf40 showing the uart3 which is the 4 coloured pins next to the S.port.
[image]

teckel12, 2018-11-12:
@Rapidlaser As I've said, you have two different issues with 2 different GPS. Focus on one thing at a time, let's focus on just the FrSky GPS. 

The FrSky GPS is connected to the S.Port. That will send GPS data to your transmitter. I don't think that sends anything to FC or INAV. So if you're expecting this GPS data to get to INAV and to the OSD, I don't think that's possible.

Once this is sorted out, we can look at the M8N issue.

KnuckleUpFPV, 2018-11-12:
https://www.rcgroups.com/forums/showthread.php?3016042-Xsrf4o-gps

KnuckleUpFPV, 2018-11-12:
You obviously missed the last comment

KnuckleUpFPV, 2018-11-12:
If you had read it. It shows the coordinates being displayed in speedybee. If they show in the configurator they will show in the osd.

Rapidlaser, 2018-11-12:
> You obviously missed the last comment

No I didn't miss it, he is using cleanflight or Betaflight and not Inav, they are not the same.

Rapidlaser, 2018-11-12:
> If you had read it. It shows the coordinates being displayed in speedybee. If they show in the configurator they will show in the osd.

Thank you but I have wired it exactly the same way, it does not show in Inav configurator, if it was that easy i would not be looking for help with a solution.

KnuckleUpFPV, 2018-11-12:
Come on man. The guy gives a link to the directions he followed. It really is that easy. You cross rx to tx, tx to rx, hook up power and ground. Turn gps for uart3 with baudrate on auto, in config tab turn on gps and pick ublox or nmea for the protocol. Leave the rest at auto. Make sure your board is powered from external power. Go near a window and go to gps tab. See if its sending packets. If not there is an issue with gps or wiring. Most times people hook this things up sitting in their house and never find any sats. Then complain their gps isnt working. All 14 of mine i hooked up I took outside and let it sit with my laptop. Then i turned on gps coordinates in osd menu saved, then went outside and let it sit powered. I just kept checking. My goggles waiting for coordinates and sat count to show.

Rapidlaser, 2018-11-12:
> Come on man. The guy gives a link to the directions he followed. It really is that easy. You cross rx to tx, tx to rx, hook up power and ground. Turn gps for uart3 with baudrate on auto, in config [...]

Please stop, I have already done this I have already told you, this does not work in Inav, he is not using Inav, if you read my original post I explain that the GPS works fine on the transmitter and picks up the sensors and shows how many satellites it picked up, there are only four wires to hook up and if I had it wired wrong it would not show on the transmitter which it does.

KnuckleUpFPV, 2018-11-12:
Inav betaflight cleanflight all basically work the same you obviously have it connected wrong if its sending it over sport and not in the fc telemetry stream.

Rapidlaser, 2018-11-12:
> Inav betaflight cleanflight all basically work the same you obviously have it connected wrong if its sending it over sport and not in the fc telemetry stream.

I don't wish to be rude but please stop posting, they do not work the same, I don't know why you refuse to accept it, as I already stated if it was wired wrong it would not send the data to the transmitter which it does, also you do not need to find satellites (even though I have taken it outside and got satellites). It will still end messages to inav if it was working, the message count is zero in inav.

digitalentity, 2018-11-12:
@Rapidlaser let's conclude the original issue "Inav with xsrf40 can read GPS telemetry on transmitter but not in inav". S.Port on INAV is used for **sending** data to receiver+radio, not for **receiving** data from other S.Port sensors. Therefore FrSky GPS connected to S.Port of XSRF40 will never work.

As for Ublox GPS connected to FC - please open another issue and provide full contect how it is connected and how firmware is configured.

closed 2018-11-12 by digitalentity

Rapidlaser, 2018-11-12:
> @Rapidlaser let's conclude the original issue "Inav with xsrf40 can read GPS telemetry on transmitter but not in inav". S.Port on INAV is used for **sending** data to receiver+radio, not for [...]

Thank you, I am not using S.port I am using UART3 as in the above mentioned posts, I think this has become clouded because I mentioned that I tried it with S.port originally before I bought the M8N and connected it to UART3, thank you for your support.

KnuckleUpFPV, 2018-11-12:
If you have an old revision of that board those pins are in fact not connected to uart3 according to frsky. You likely have an old version if all is correct. This is direct from frsky. They say they will include uart3 functionality on the next revision of the board.

digitalentity, 2018-11-12:
Remember to open a separate issue for one problem. This helps avoid confusion.

KnuckleUpFPV, 2018-11-12:
https://www.rcgroups.com/forums/showthread.php?2902661-4-flight-controllers-integrate-with-FrSky-XSR-receiver-release!

For reference. Check with a multimeter for continuity from the pins on the chip to the pad. Guy in this forum soldered straight to the chip.

Rapidlaser, 2018-11-12:
> Remember to open a separate issue for one problem. This helps avoid confusion.

Thank you.