iNavFlight/inav #4015: Can't properly arm Inav 2.0.0 from hands

Improg, 2018-11-15:
## Current Behavior
When I arm quadcopter situated on the ground => all is OK. When I arm quad when hold it in my hands, with some great probability (more than 50%) after arming it run motors on one side (usually right) with maximum throttle. Real throttle is at minimum position of course. Quad tries to rotate in hands, but it doesn't try to self align in some position, it can rotate by 90 degrees and will continue run motors at full throttle chaotically. Disarm is needed to avoid crush.
In other cases (less then 50%) it run all motors slow according to the minimum throttle value as expected, so I can give throttle more and take of normally.
ACRO/AIR mode. As I remember, the same behavior in ANGLE mode.
Inav 1.9.1 has no such bug.
It is usefull to take off from hand for me because of action camera mounted to bottom, the quad has no "legs" so uncomfortable situate is horisontally on the ground.

## Steps to Reproduce
1. Take a quad in a hand;
2. Align it normally, so motors will be directed vertically to the sky, similar to position when it lies on the ground;
3. Arm.

## Expected behavior
Run all motors as expected according to minimum throttle value.

- FC Board name and vendor:
Omnibus F4 Pro V3 not clone, from Airbot
- INAV version string:
2.0.0

wx4cb, 2018-11-15:
if it's uncomfortable to launch because of the camera underneath, how do you land it ? the other thing is, do you have it in air mode? that could cause it as it thinks it's not level and will try to compensate.

Improg, 2018-11-16:
I land it in POS HOLD in hands, or in air mode in grass. Because of camera underneath it have big angle when lies on the ground, so it can't rotate two props to take of, but it can be landed.
Yes, this behaviour is in air mode/acro. If it will try to compensate level it will reduce motor power during rotation, but it didn't.

wx4cb, 2018-11-16:
air mode and acro are two different things. try turning off air mode and see what happens

closed 2019-01-30 by stale[bot]