iNavFlight/inav #4018: Matek F722-STD - OSD fill display with black dashed lines

CradyLab, 2018-11-16:
## Current Behavior

[image]

## Steps to Reproduce

1.  Flash FC with latest release of iNav
2.  Enable OSD in configurator - no need to configure further - bug shows up immediately
3.  To eliminate anomaly disable OSD
4.

## Expected behavior

Enabling the OSD feature fills the displayed video with black horizontal dashed lines from top to bottom.  Characters and camera feed are still visible - the lines are effectively an overlay. 

As a test I reflashed the same FC with Butterflight and enabled OSD - no issues.  Display was clean and error-free in Butterflight.

## Suggested solution(s)

## Additional context

---

- FC Board name and vendor:

Matek Systems F722-STD
http://www.mateksys.com/?portfolio=f722-std

- INAV version string:

# version
# INAV/MATEKF722 2.0.1 Oct 31 2018 / 19:27:11 (a04e68aaa)

shellixyz, 2018-11-16:
You have to update the OSD fonts: up to date configurator's OSD tab -> upload fonts

closed 2018-11-20 by DzikuVx

CradyLab, 2018-12-03:
> You have to update the OSD fonts: up to date configurator's OSD tab -> upload fonts

Doesn't work.  Attempts to upload fonts forces the board to reboot

giacomo892, 2018-12-03:
Please @CradyLab don't spam the issue tracker commenting on closed issues and in the meanwhile opening new ones regarding the same problematic.