iNavFlight/inav #4019: Access BlackBox SDCard from computer (Matek F722-STD)

Frontier314, 2018-11-17:
## Current Behavior
Accessing logs on the installed SD card on a flight controller like Matek F722-STD, requires removing the card from the slot. This is inconvinient for builds where the flight controller is inside a shell (i.e like this hexacopter: https://www.thingiverse.com/make:554274).

## Desired Behavior
To access the contents of the SD Card from the computer, like Betaflight 3.5 does.

## Suggested Solution
Cherry-pick (if possible) the SD card code from Betaflight 3.5.

## Who does this impact? Who is this for?
All users.

Thank you.

wx4cb, 2018-11-17:
Same thing applies for the Matek 405 Wing controller as it's on the bottom of the board.

flyfisher2, 2018-11-18:
It is actually a problem with all FCs, regardless of where the reader is mounted.
It has been reported numerous times in the past but obviously not considered important by the devs.
Apart from the access problem, the reader/chip is not going to last very long if it is removed and inserted after every flying session.
Currently the only solution is to switch to Betaflight or Ardupilot.

shellixyz, 2018-11-19:
Duplicate of #1336

shellixyz, 2018-11-19:
Closing since there is another issue for this feature request

closed 2018-11-19 by shellixyz