iNavFlight/inav #4022: Failsafe during launch mode.

xermalk, 2018-11-18:
## Current Behavior
Inav will continue to maintain autolaunch, if a failsafe is triggerd. This is really bad as on the outside everything looks normal. And allows the plane to get into some really bad positions before triggering the failsafe.

Video
https://www.youtube.com/watch?v=Hje7MY5PCOw
Blackbox
https://ufile.io/zsrze

## Steps to Reproduce
Get a failsafe in launch mode

## Expected behavior
Failsafe triggering

## Suggested solution(s)
Prioritize failsafe over launch mode.

## Additional context
https://pastebin.com/d6snsQ3V

---
- FC Board name and vendor: INAV/MATEKF405SE 
- INAV version string: 2.0.1 Oct 31 2018 / 19:28:11 (a04e68aaa)

shellixyz, 2018-11-18:
It could be dangerous to trigger RTH very low to the ground but you had enough sats and low enough EPH it should have triggered after the launch timeout or altitude threshold was reached. Looks like a bug to me.

kardon18, 2018-11-18:
Was it a physical disconnection of the RX from the FC? 
Looking at your video. The FS occurred when the motor started. Before you launched it. 
I wonder if it was a combination of things that lead to this.

FS occurred with in **set failsafe_min_distance = 2000**.  Because it was there at the point of launch.
Then because - **set failsafe_min_distance_procedure = DROP**  
Soon as the FC exited the Auto Launch sub, it seemed to back tracked to the point at which the FS occurred, which was within **failsafe_min_distance** and initiated the DROP procedure and disarmed by Failsafe.

@xermalk  _Prioritize failsafe over launch mode_.   

That seems like the fix.

wx4cb, 2018-11-18:
my only concern would be as mentioned before by shellixyz. now obviously, usually you're not launching towards obstacles (unless you fly where i fly and have no choice), but maybe you can - in this situation - override the "climb before rth" setting to make sure that it climbs out first to lets say... a minimum RTH altitude that can be set. so lets say youre RTH altitude is 400 feet, but the minimum you can go is say 150, then make sure it lcimbs out to that before turning? 

just my thought... but override launch with FS can be tricky, do you just want it to abort the launch and glide down in a straight line or trigger failsafe completely?

shellixyz, 2018-11-18:
Indeed I didn't see that `failsafe_min_distance` had been set. That's most likely why it didn't RTH. Good catch @kardon18.

Prioritizing failsafe over launch mode would make sense in this case. Maybe prioritizing failsafe if the failsafe procedure is DROP I think would be the solution. I'm still convinced that prioritizing FS when the action would be to RTH just after launch would be a bad idea.

kardon18, 2018-11-18:
@shellixyz   _Maybe prioritizing failsafe if the failsafe procedure is DROP I think would be the solution._

I agree. That does sound like a specific way of going about it.  

I also understand your concern regarding an instant entry in RTH after launch.

How much of a concern it poses, will depend on what - **set nav_rth_alt_mode** that is selected.
e.g. **set nav_rth_alt_mode = EXTRA or AT_LEAST** should not pose a problem at all. While some of the others would.

As @wx4cb stated -  
 _in this situation - **override the "climb before rth**" setting to make sure that it climbs out first to lets say... a minimum RTH altitude that can be set. so lets say youre RTH altitude is 400 feet, but the minimum you can go is say 150, then make sure it lcimbs out to that before turning_

wx4cb, 2018-11-19:
The problem i foresee is there's some places i fly where having it turn before climb would be a BAD idea at low altitude :D planes and windows don't mix heh

xermalk, 2018-11-19:
What about hard coding it to go to drop, if it enters failsafe early in auto launch.
As to why it happened. I'm pretty sure i accidentally pulled a bit on the power cable, when i had to tilt the fc to get to the sd card port.

That's something i'm sorting out in the rebuild.

shellixyz, 2019-03-02:
Looks like a duplicate of #2267

digitalentity, 2019-06-10:
This only seems to affect setups with `failsafe_procedure=RTH`.

Possible solutions:

1. Allow failsafe to abort launch mode and switch to RTH
2. Override failsafe procedure for launch mode to `DROP` and `LAND`

What should be implement?

digitalentity, 2019-06-10:
Naive fix (allow failsafe to abort launch mode) - #4819

Dronek, 2019-06-14:
> This only seems to affect setups with `failsafe_procedure=RTH`.
> 
> Possible solutions:
> 
> 1. Allow failsafe to abort launch mode and switch to RTH
> 2. Override failsafe procedure for launch mode to `DROP` and `LAND`
> 
> What should be implement?

Override failsafe procedure for launch mode to `DROP` with no further autopiloting activity would be the most practible fix IMHO.

closed 2019-06-18 by digitalentity