iNavFlight/inav #4025: Support magnetometer connected to SDA1 / SDC1 on HAKRC-MiniF4

alexbirkett, 2018-11-19:
## Current Behavior

The [HAKRC-MiniF4](https://www.banggood.com/20x20mm-HAKRC-MiniF4-Flytower-F4-Flight-Controller-AIO-OSD-BEC-20A-BLheli_S-ESC-for-RC-Drone-p-1246529.html?rmmds=search&cur_warehouse=CN) is one of only a 20x20mm flight controllers with built in barometer. INAV is currently supported using the `OMNIBUSF4V3` target.

The HAKRC-MiniF4 has a JST connector that has a UART and a i2c pins (SCL1 and SDA1) broken out on it that could possibly be used to connect a GPS / magnetometer module e.g Beitan BN-880 .The [instructions]((https://www.banggood.com/20x20mm-HAKRC-MiniF4-Flytower-F4-Flight-Controller-AIO-OSD-BEC-20A-BLheli_S-ESC-for-RC-Drone-p-1246529.html?rmmds=search&cur_warehouse=CN)) suggest the magnetometer is connected to SCL2 and SDA2 which means some messy soldering and using up pads that could be used for something else:

[image]

## Desired Behavior
Make it possible to connect a magnetometer to SC1 / SDA1 on the HAKRC-MiniF4.

## Suggested Solution
Create a new target specifically for the HAKRC-MiniF4 that uses `USE_I2C_DEVICE_1`.

## Who does this impact? Who is this for?
People using HAKRC-MiniF4 with external GPS / magnetometer module.

wx4cb, 2018-11-19:
that looks like a cool little board combo. got links to the parts you're using ? I might use it in a new nano goblin i'm doing

alexbirkett, 2018-11-19:
I've tried various permutations and combinations of the following to make it possible to connect the magnetometer to SCL1 and SDA1:

* Remove `#define USE_I2C_DEVICE_2` and add `#define USE_I2C_DEVICE_1`
* Disable features that use `PB6` and `PB7`
* Remove  `#define MAG_I2C_BUS             BUS_I2C2` and add `#define MAG_I2C_BUS             BUS_I2C1`

None of this worked. Could there be some extra circuitry connected to SCL2 and SDA2 that is not connected to SDA1 / SCL1? There appears to be an open circuit from the pads to the pins on the IC. How can I debug this?

Changes here:
https://github.com/alexbirkett/inav/commit/a7c63d5a4bc39b57ae5c5597fb3c45735fa7b485

This is the second time that I've tried and failed to remap resources. I'd really like to figure out what I'm doing wrong.

alexbirkett, 2018-11-20:
@atobiese 
`#define I2C1_SCL PB6
#define I2C1_SDA PB7` fixes it! Thanks.

alexbirkett, 2018-11-21:
This issue is not fixed @shellixyz. I hope I can submit a PR that fixes it.

msfujino, 2019-05-18:
I modified and remapped HAKRC MiniF4.
I think this information will be helpful for you.

[image]
[image]
[image]

chobitsfan, 2019-07-23:
hi @msfujino 
May I ask a question? Do I need to physically rewire PA1 to LED? Or you mean it has PA1 has been rewired to LED when board come out? Thank you very much
[image]

msfujino, 2019-07-23:
Hi chobitsfan

Because I wanted to connect WS2811_LEDStrip to this LED terminal, I cut the pattern, jumpered the PA1 and LED terminal, and assigned PA1 to WS2811 drive signal. 
If you do not use this LED terminal, you do not need pattern cutting and jumper wiring.

msfujino

chobitsfan, 2019-07-23:
hi @msfujino thank you
You do rewiring because originally the LED terminal in hakrc mini f4 is connected with SCL. Is it correct? thank you

msfujino, 2019-07-23:
Hi chobitsfan
The resistance between PB6-LED-SCL is almost zero. The same waveform as PB6 is observed with LED and SCL. Although I did not see the pattern directly, I think that PB6-LED-SCL is connected.
msfujino

chobitsfan, 2019-07-23:
hi @msfujino thank you very much. I got it
It is strange that hakrc connected LED and SCL together.  It do not seems make sense?

Ex4nder, 2020-04-07:
Thank you very much @msfujino, your post solved my problems both in Betaflight and Inav

msfujino, 2020-04-07:
Hi Ex4nder, I am glad to help you.
It is difficult to solder the jumper directly to PA1(pin 15), so I use PA0(pin 14) for LEDstrip by soldering the jumper to RSSI pad that I do not use.

Ex4nder, 2020-04-08:
I didn't configure LED pin yet, and right now I've an error I can't arm because of invalid LED setting, but I will like more to enable hardware serial port 4 on PA0/PA1 for lidar and optical flow, and use maybe beeper pin for the led, beeping with ESCs. I still have to read well the datasheet to see if it can be done. About RSSI I also think I don't need it, because I've telemetry on fport (serial 1 RX in half duplex)

msfujino, 2020-04-09:
The ESC tower connector has 12 pins, but 2 pins each are connected together (6 pairs).
Since it is used for VBAT, GND, S1, S2, S3, and S4, I don't think it is possible to remap other functions.

Ex4nder, 2020-04-09:
No, it's not, I also checked yesterday night.
In the end I have working for now:
Serial TX1 as f.port
Serial TX3, RX3, Benewake rangefinder
Serial TX4 (rssi pin) Smart audio VTX control
Serial TX6 BN880 GPS
I2C_1 (PB7,PB6) BN880 Magnetometer

I will go on in the next days in connecting LED ad Buzzer, and if everyting will work I will report back.

msfujino, 2020-04-12:
It took time to set up the compilation environment, but I remap and use FCs which are not supported by iNav. The betaflight target files are very helpful.

closed 2020-12-11 by DzikuVx